MichaelGrupp/evo #282: Custom plotting of trajectories / curves based on scalar type collections

YoshuaNava, 2020-06-09:
**Is your feature request related to a problem? Please describe.**
Currently evo is able to plot APE, RPE, and trajectories, either full or split between the multiple degrees of freedom that form the path.

**Describe the solution you'd like**
It would be nice to plot trajectories and color them based on custom scalar type collections. For example, a matching score, processing time, etc.

Walking the path towards the robust perception age, this would help in finding correlations between different metrics that are not poses, but that actually reflect the behavior of your system during runtime.

**Describe alternatives you've considered**
None yet, I don't think there are other tools as flexible as evo. The nice workflow of evo when it comes to report generation was the seed for this idea.

I would like to propose this idea before trying to implement it myself in a fork or something else.

**Additional context**
Nothing in particular.

MichaelGrupp, 2020-06-12:
The plot module of evo has a function for doing that, it is the same thing that is used in `evo_ape` and `evo_rpe`. You can use any scalar array for colormapping the edges between poses, as long as the length of the array has the same length as the trajectory: https://github.com/MichaelGrupp/evo/blob/63cdfc8cbbb0a388272ce48bc2fcba6af48e3adb/evo/tools/plot.py#L366

Example script that uses this:
https://github.com/MichaelGrupp/evo/blob/master/doc/examples/custom_app.py

---

> It would be nice to have an extension to specify CSV files that contain values of a time series that we would use to color our curves

A script that does this for a single trajectory shouldn't be difficult to implement. It could also go to the scripts folder [contrib/](https://github.com/MichaelGrupp/evo/tree/master/contrib).

I'm not sure yet if/how this is something that should be integrated in the default command line apps, it be a bit complicated because `evo_traj` supports 1...n trajectories. Having to specify a CSV file for each of them is probably not so useful.

So right now I would say you can do this in custom code and maybe contribute the script in a PR if it's useful for more people.

YoshuaNava, 2020-06-28:
I will test this out locally, and it it goes well I'll submit a PR. Thank you!

closed 2020-06-29 by MichaelGrupp