MichaelGrupp/evo #285: Current limits of the tool

YoshuaNava, 2020-06-11:
It would be interesting to know what are the current limits of the tool, and how to circumvent them. For example:

* How many files can be plotted at the same time by the tools `evo_res` and `evo_traj`?
* What if different files have different number of poses? How does that affect the resulting reports?

MichaelGrupp, 2020-06-12:
> How many files can be plotted at the same time by the tools evo_res and evo_traj?

There's no limit actually. But if you have a gazillion files, it might start to get a bit slow and messy at some point.

> how to circumvent

Pretty much depends on the limits you encounter.

> What if different files have different number of poses? How does that affect the resulting reports?

There's no rule of thumb for that, it depends on a lot of factors. Generally speaking it can make a difference and it's up to the user to decide if the comparisons still make sense or not for that particular dataset.

YoshuaNava, 2020-06-28:
Have you considered the possibility of outputting Jupyter notebooks? Where users can interact with the plots, and visualize results in sort of a global way?

I ask this because SLAM is in many cases non-deterministic, and for proper characterization of an algorithm in a certain environment, running it more than once is a good way. Then, comparing results from different runs on different environments / with different params would be useful.

MichaelGrupp, 2020-06-29:
I don't completely understand what you mean, but there are some example notebooks in `notebooks/`. I don't plan to extend these Jupyter notebooks in the near future, but good contributions would be welcome.

> gathering a good amount of execution samples is a good way. Then, comparing results from different runs on different environments / with different params would be useful

How is that related to Jupyter notebooks?

MichaelGrupp, 2020-08-20:
Closing this for now, feel free to reopen if you have concrete questions or ideas.

closed 2020-08-20 by MichaelGrupp