MichaelGrupp/evo #286: Wrong number of max. possible matching timestamps

jcremona, 2020-06-11:
**Description:**
When an alignment between two trajectories is produced (-a option), the displayed number of maximum possible matching timestamps is wrong. 
https://github.com/MichaelGrupp/evo/blob/5987ec3f39a2b73b034b5d2d7ae0e2a9e1c432df/evo/core/sync.py#L78
https://github.com/MichaelGrupp/evo/blob/5987ec3f39a2b73b034b5d2d7ae0e2a9e1c432df/evo/core/sync.py#L101-L106

IMO, it should be the length of the shortest trajectory.

**Command:**
```
evo_traj tum --ref sequence01_gt.txt sequence01_ests.txt -p -a -v
```

**Console output:**
```
--------------------------------------------------------------------------------
Loaded 4817 stamps and poses from: sequence01_ests.txt
Loaded 1392 stamps and poses from: sequence01_gt.txt
--------------------------------------------------------------------------------
Found 416 of max. 4817 possible matching timestamps between...
	reference
and:	sequence01_ests.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------

```

Please give also the following information:
* evo version: v1.7.2
* Python version: 3.8.2
* operating system and version: Ubuntu 20.04
* did you change the source code?: no 
* output of `evo_config show --brief --no_color`: 
```
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_unknown_cell_value": 205,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true
}

```

MichaelGrupp, 2020-06-12:
Correct. Do you want to send a pull request that changes this line?

jcremona, 2020-06-12:
@MichaelGrupp I'll send the pull request, thanks for your answer.

closed 2020-06-12 by MichaelGrupp