MichaelGrupp/evo #291: Installation failure

buemi, 2020-06-17:
**Description:**

Hi, I'm trying to install evo in Anaconda environment using the following command:

python -m pip install evo

but I receive the following error:
_Requirement already satisfied: evo in c:\users\myname\evo (1.9.1)
Requirement already satisfied: numpy in c:\users\myname\anaconda3\envs\deeplearning\lib\site-packages (from evo) (1.18.1)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x00000267188FDF08>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))': /simple/matplotlib/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x000002671891FCC8>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))': /simple/matplotlib/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x000002671891FF88>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))': /simple/matplotlib/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000026718924988>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))': /simple/matplotlib/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000026718924EC8>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))': /simple/matplotlib/
ERROR: Could not find a version that satisfies the requirement matplotlib (from evo) (from versions: none)
ERROR: No matching distribution found for matplotlib (from evo)

* operating system and version: Windows 10

MichaelGrupp, 2020-06-17:
Are you maybe connected a corporate network that has a proxy server? The output shows problems related to proxy.

Maybe the proxy options of pip could help here: https://pip.pypa.io/en/stable/user_guide/#using-a-proxy-server

buemi, 2020-06-17:
Oh, yes, sorry for your time. I used the wrong command to set the HTTPS env variables. The correct command is : 
conda env config vars set HTTPS_PROXY=http://user:passwd@proxy:server:port/
Than I successfully installed evo.
Thank you for your help.

closed 2020-06-17 by buemi