MichaelGrupp/evo #295: How to solve the error that the roll and yaw angles jump between -pi and pi

zzzzzzli, 2020-06-24:
**Is your feature request related to a problem? Please describe.**
I use `evo_traj` tool to plot the euler angle ground truth of Euroc v101 and kitti dataset, and the results shows that the yaw and roll angles jump bewteen -pi and pi.

**Describe the solution you'd like**
The yaw and roll angle keeps positive or negative near their real value until they really changes.

MichaelGrupp, 2020-06-24:
What do you expect instead?

zzzzzzli, 2020-06-25:
to make the data more stable is possible?

MichaelGrupp, 2020-06-25:
The angles are defined between [-pi, pi] (right hand rule), so at some point jumps can happen. I don't know what you mean with "real value" but this is how it is calculated. What you could try is to change the axis order if you need to via `evo_config set euler_angle_sequence sxyz`, where you would need to replace "sxyz" by the one you want (explained here: https://github.com/MichaelGrupp/evo/blob/master/evo/core/transformations.py#L87).

zzzzzzli, 2020-06-25:
Ok, thank you very much

closed 2020-06-28 by MichaelGrupp