MichaelGrupp/evo #301: What exactly `--all_pairs` do with RPE ?

mcamurri, 2020-07-03:
Hi @MichaelGrupp , excellent work!

I have a technical question on the `--all_pairs` option of `evo_rpe`. 
Let's say I have a ground truth signal and an estimated signal, both already aligned and of the same frequency. 

I normally set a distance of `10 m` as my delta parameter for the RPE.

For simplicity, let's pretend the robot has a constant velocity magnitude of 1 m/s and the samples are at 1 Hz, such that at the time `t5` correponds a travelled distance of 5 m (in ground truth).

What I understood is that:
- **without `--all_pairs` option:** `evo_rpe` computes the error between time `t0` and `t10`, then it re-aligns the two trajectories from time `t10` onward, it computes the error at time `t20`, re-aligns at time `t20` and so on. This means that for a 500 m long trajectory only 49 RPE samples are computed. These are aggregated to produce the final table with mean, median, standard deviation, etc.

- **with `--all_pairs`**:  `evo_rpe` computes the error between time `t0` and `t10` as before. Then, it aligns the two trajectories at time `t1` and computes the error between `t1` and `t11`, then from `t2` and `t12` etc. This way all the samples from time `t10` onward are used, giving a total of 490 RPE samples, again aggregated to get the final results.

Is my interpretation correct? 

On a more complicated case: if the two trajectories don't have the same frequency, would `evo_rpe` use the ground truth one as reference, looking for the closest in time on the estimated trajectory to compute the error?

MichaelGrupp, 2020-07-08:
Your understanding looks good. What do you mean with re-aligning?

> if the two trajectories don't have the same frequency, would evo_rpe use the ground truth one as reference, looking for the closest in time on the estimated trajectory to compute the error?

* trajectories are synced to have matching nr. of poses based on timestamps (by default in both `evo_ape`  & `evo_ape`)
* we look for all pose pairs in the estimated trajectory that have `<delta>` distance
* the RPE is calculated for these pairs using the time-wise matching poses from the groundtruth

In the last two points you could swap estimate and groundtruth, it gives slightly different results (you can try by swapping them in the command line, too).

mcamurri, 2020-07-08:
From the most commonly used paper:
[image]

I'll try to draw it:
[image]

The inversion of Q_i and P_i inside the brackets is equivalent to aligning the initial poses P_i with Q_i and then compute the green segment. This is done for every red segment.

Can you confirm then that with `--all-pairs` we have 490 rpe samples in the toy examples I explained above?

If that is the case, why is it not possible to plot it with the colorscale? 

If I don't use all pairs, for a trajectory of 240 meters I end up with only a few  pieces of straight lines.

Before using `evo` I used to do this with matlab (which I want to abandon) and I was plotting basically the same plot as your color map one, but for all the samples except the first 10 m of trajectory.

MichaelGrupp, 2020-07-08:
Ok, all fine. I was just confused by the term alignment here.

> Can you confirm then that with --all-pairs we have 490 rpe samples in the toy examples I explained above?

Yes, this should be the case.

> If that is the case, why is it not possible to plot it with the colorscale?

It's not impossible, it was just not implemented and thus I throw that error. Main reason was that I was not convinced if a colormap plot can be intuitively interpreted in the all_pairs case.

I just removed that error in a test branch, so feel free to try it and see if the plot looks good to you: https://github.com/MichaelGrupp/evo/tree/issue-301

mcamurri, 2020-07-08:
Thank you @MichaelGrupp , that clarifies my questions. Also, thanks for the fix. 
I wasn't sure if the error was due to a missing feature or if it was mathematically impossible to achieve. 
The issue can be closed.

MichaelGrupp, 2020-07-08:
Ok, cool. I would then test a bit more with the change I made and then probably merge it to master if it doesn't cause any errors. Feel free to try it, too. PR: #303

closed 2020-07-08 by MichaelGrupp