MichaelGrupp/evo #302: Questions about Umeyama Alignment and Evaluation Results

zxl19, 2020-07-05:
Hi Michael, thank you for sharing your work.
I'm using evo on Windows 10 with Python 3.8 and recently encountered this problem.

# Scenario

I'm using the rosbag I recorded to test A-LOAM. The algorithm outputs two pose estimates at different frequencies(~1Hz and ~10Hz), I use GPS and IMU information(~100Hz) as groundtruth to evaluate the algorithm. **First I synchronized the timestamps to get the same amount of poses for evaluation. Then I transformed longitude and latitude into UTM coordinates and converts them into relative pose. Next I exported the poses into KITTI format. Note that the odometry poses are in LiDAR coordinates and the (relative) GPS poses are in UTM coordinates.**

# Questions

## About evaluation metrics

Since the groundtruth and the odometry results are in different coordinates, and **the extrinsics between LiDAR in the first frame and UTM is hard to get**, I used the following command and got the results as follows:

```shell
evo_traj kitti groundtruth.txt laser_odometry.txt integrated.txt after_mapped.txt -p
```

[image]

**In this scenario**, are there any ways to **evaluate the results using the metrics provided by KITTI benchmark**?

## About Umeyama alignment

Later I aligned the poses using Umeyama alignment and in some cases the results looks fine:

[image]

But when I was evaluating hdl_graph_slam using the command below, I got the following result:

```shell
evo_ape kitti groundtruth.txt odometry.txt -va -p --plot_mode xy
```

(P.S. Is `-va` the same as `-a`? I didn't see this suffix in Wiki.)

[image]
**The start of the two trajectories(the bottom right corner) did not even match, which makes the result unreliable.** Are there any ways to solve this problem(like **aligning the trajectories while fixing the start points**)? 

**Also, how should we treat the aligned results? When the extrinsics are unkown, can we consider the aligned results as the best or the best possible result?**

## About map quality metrics

My question is pretty much the same as #280,. **My understanding is that what evo and KITTI Odometry does is evaluating the precision of odometry(relative trajectories) instead of the precision of the map itself.** Then it's pretty hard to evaluate how accurate the map is, especially in autonomous driving applications. **So far, do you know if there are any methods to evaluate the precision of the map?**

# Summary

Thank you again for your excellent work. I understand that some of my questions are hardly related to evo itself, but it means so much if you could help me with those problems.

Sincerely.

MichaelGrupp, 2020-07-08:
> The start of the two trajectories(the bottom right corner) did not even match

This is not a criteria for Umeyama alignment. It tries to align all the points of the trajectories such that they overall fit the best possible way with a rigid transformation (`--align`) and/or a scale correction (`--correct_scale`)

> can we consider the aligned results as the best or the best possible result?

:thinking:... alignment methods can reduce the effect of things like offsets, scale issues...

> aligning the trajectories while fixing the start points

You can try `--align_origin` instead, it glues the first pose of the estimate trajectory to the first (time-wise matching) reference pose. But it can't be mixed with Umeyama alignment and will show you drift from start to end.

---

> My understanding is that what evo and KITTI Odometry does is evaluating the precision of odometry(relative trajectories) instead of the precision of the map itself.

Usually there is a direct correspondence between the trajectory and the "map". Besides that, I would stick to my comment here: https://github.com/MichaelGrupp/evo/issues/280#issuecomment-643274818

---

> Is -va the same as -a? I didn't see this suffix in Wiki

`-va` is a shortcut for `-v -a`, which is again a shortcut for `--verbose --align`. Hope this helps.

zxl19, 2020-07-09:
Thank you for your reply!

> You can try --align_origin instead, it glues the first pose of the estimate trajectory to the first (time-wise matching) reference pose. But it can't be mixed with Umeyama alignment and will show you drift from start to end.

I have tried this command suffix before using Umeyama alignment, the command and results are shown as follows:

```shell
evo_ape kitti groundtruth.txt odometry.txt -p
```

[image]

```shell
evo_ape kitti groundtruth.txt odometry.txt --align_origin -p
```

[image]

By using `--align_origin` I glued the start of the trajectories together but **somehow rotated** the trajectory. Also, in those two cases the odometry results doesn't really 'match' the groundtruth. I think it is because the extrinsics between groundtruth and odometry are missing or not calibrated since I'm using my own data.

> **(In the original question)**
> # Scenario
> I'm using the rosbag I recorded to test A-LOAM. The algorithm outputs two pose estimates at different frequencies(~1Hz and ~10Hz), I use GPS and IMU information(~100Hz) as groundtruth to evaluate the algorithm. First I synchronized the timestamps to get the same amount of poses for evaluation. Then I transformed longitude and latitude into UTM coordinates and converts them into relative pose. Next I exported the poses into KITTI format. Note that the odometry poses are in LiDAR coordinates and the (relative) GPS poses are in UTM coordinates.

In this case, is it possible to evaluate the precision of the algorithm?

------

> This is not a criteria for Umeyama alignment. It tries to align all the points of the trajectories such that they overall fit the best possible way with a rigid transformation (--align) and/or a scale correction (--correct_scale)

I'm reading the four papers you mentioned in `metrics_tutorial.ipynb` and understanding the algorithm. Since I'm using LiDAR SLAM, there is no need to correct the scale. **I'm wondering if it's possible to offset the effect of missing extrinsics(after all it's also a rigid transformation) by using Umeyama alignment?**

In practice, I found the result after Umeyama alignment **sometimes counterintuitive**.

> [image]

The groundtruth and the odometry results I used are in relative poses(starting at `(0, 0, 0)`). Since the drift of SLAM algorithms increases over time, I think the error should not be that much at the begining of the trajectories, even if it's not a criteria for Umeyama alignment. It's a little bit counterintuitive and confuses me a lot these days.

Best regards.

zxl19, 2020-07-09:
Also, I have a couple of questions about the `tum` format with the `--sync` command suffix. How is the sychronization done? Does it require the timestamps to be exactly the same or should the user manually set a tolerance? Or are there any interpolation methods used?

Thanks.

MichaelGrupp, 2020-07-09:
> glued the start of the trajectories together but somehow rotated the trajectory

Then they probably have pose axes with different orientations. You might see what's up by enabling axis markers: https://github.com/MichaelGrupp/evo/wiki/Plotting#coordinate-axis-markers

> I'm wondering if it's possible to offset the effect of missing extrinsics(after all it's also a rigid transformation) by using Umeyama alignment?

No. If your trajectories have been captured by different sensors with offsets between each other, they can have different shapes. You need to account for that yourself by transforming the poses according to the calibration.

> I think the error should not be that much at the begining of the trajectories

The APE metric with alignment shows how well the overall fit of two trajectories is. Whether the alignment creates a gap at the beginning or not doesn't matter, it's the best possible fit. If this is not what you expect, you need to transform the trajectories in another way, or use the relative error which doesn't care about global offsets.

>  How is the sychronization done? Does it require the timestamps to be exactly the same or should the user manually set a tolerance? Or are there any interpolation methods used?

https://github.com/MichaelGrupp/evo/wiki/evo_traj#temporal-alignment
No interpolation.

zxl19, 2020-07-13:
Thank you, the problem is solved. Closing this issue now.

7zhoulei, 2020-08-14:
[@zxl19 hello, i also have the same problem as you. i want to evaluate a lidar slam performance. But when i use --align_origin, the estimated trajectory does not really match the ground truth. The estimated trajectory is rotated. Could you please tell me how to make the estimated trajectory match the ground truth? thx.

[image]

zxl19, 2020-08-15:
Hi @7zhoulei 

By using `--align_origin`, you can only glue the starts of two trajectories together. However, the extrinsic between your ground truth and LiDAR odometry result is unknown and remains to be calibrated. There are two ways to solve this problem.
1. Use `-va` instead of `--align_origin`, evo will use Umeyama alignment to align those two trajectories and give you a SE(3) transformation matrix.
The advantage of this method is it's simple, but sometimes the result after alignment is counterintuitive. You may refer to [my comment above](https://github.com/MichaelGrupp/evo/issues/302#issuecomment-655878704) for more details.
2. Calibrate the extrinsic between your LiDAR and the sensor which you used to acquire ground truth(e.g. an inertial and GPS navigation system) and add a `.json` file.
For specific instructions, you can refer to [this chapter](https://github.com/MichaelGrupp/evo/wiki/evo_traj#alignment-and-transformation) in evo Wiki.

I hope my answer will be of help. Also, it would be wonderful if you could upload your result after trying the methods above. I'm also bugged by this problem recently and would like to see the results in different scenarios.

Best wishes.

7zhoulei, 2020-08-15:
@zxl19 thanks for your reply. 
1. the first method (use -va instead of --align_origin) can align the estimated trajectory with the ground truth(see figure1). But the alignment is not what I want. "-a" uses Umeyama alignment which tries to align all the points of the trajectory, and leads to the best possible overall fit. In my opinion, at the start position the error should be zero. And I want to see how the error grows over driving distance or time. Unfortunately, Umeyama alignment can not achieve this goal. I just want to align the first several points of the estimated trajectory with the ground truth. 
[image] fig. 1
2. for the second method, I use "--align --n_to_align 20 -v" to align the first 20 poses of the estimated trajectory with the ground truth. (see Fig. 2). which align  the estimated trajectory with the ground truth as I expected. It also shows me the transformation matrix between the estimated trajectory and the ground truth. But "--align --n_to_align 20" just works for plotting the trajectory. When I use it to compute RPE or APE, it doesn't work.
I also tried to use **.json** as described in [https://github.com/MichaelGrupp/evo/wiki/evo_traj#alignment-and-transformation](url), it works for plotting, but I don't know how to use **.json** or "--align --n_to_align 20 " to align the estimated trajectory for computing the error. Or it is impossible to use **.json** or "--align --n_to_align 20 " for computing the error.
[image] fig.2

zxl19, 2020-08-17:
Hi @7zhoulei 

> But the alignment is not what I want. "-a" uses Umeyama alignment which tries to align all the points of the trajectory, and leads to the best possible overall fit. In my opinion, at the start position the error should be zero.

That is also what I found counterintuitive. Since the drift of SLAM algorithms increases over time, the error should be small at the beginning.

> Or it is impossible to use .json or "--align --n_to_align 20 " for computing the error.

You are right, it seems that it is not supported in `evo_ape` or `evo_rpe`, here is what I got:

```shell
evo_ape: error: unrecognized arguments: --n_to_align 20
evo_ape: error: unrecognized arguments: --transform_left extrinsics.json
```
> I use "--align --n_to_align 20 -v" to align the first 20 poses of the estimated trajectory with the ground truth. (see Fig. 2). which align the estimated trajectory with the ground truth as I [...]

I haven't tried `--n_to_align` before, but I think since you can get the transformation matrix by aligning the first 20 points in the trajectory by using `evo_traj`. You can then use the transformation matrix you got to manually align the trajectories before using `evo_ape`. The result should be better than that by using `-va`. I will also try this method later and I'll let you know my results.

Also, I'm reopening this issue to get more advice from others.

Best regards.

MichaelGrupp, 2020-08-17:
You can save the aligned/transformed trajectory from `evo_traj` with `--save_as_*` and then use this new trajectory for evo_rpe/ape. Besides that it's probably worth to add `--n_to_align` also to evo_ape / evo_rpe by adding a CLI flag for the [method](https://github.com/MichaelGrupp/evo/blob/819720d91e652acffe9dc5b64d1d1172dbc61895/evo/core/trajectory.py#L370).

> In my opinion, at the start position the error should be zero. And I want to see how the error grows over driving distance or time.

Even if you do `--align --n_to_align X`, you still get an Umeyama alignment of these first X points, so there's no guarantee that your start is "zero". The only way to guarantee this is origin alignment with `--align_origin`. But for this, the trajectories must be in the same tracking frame, so you need extrinsic calibration as discussed before.

7zhoulei, 2020-08-17:
Hi Michael, thanks for your reply. Your answer is really useful.  ```evo_traj``` with ```--align --n_to_align xxx ``` and ```--save_as_*``` can save the aligned trajectory as man expected automatically. I don't need to transform the unaligned trajectory using the transformation matrix got by  ```evo_traj``` with ```--align --n_to_align xxx -v```  manually. 

> In my opinion, at the start position the error should be zero. And I want to see how the error grows over driving distance or time.

Sorry for the inaccurate expression.  Actually, I want to express that the errors at the first several points of the estimated trajectory should be small. And the errors should increase over time. When I just align the first several points, I can see how the errors change over time/distance. Now my problem is solved.

zxl19, 2020-08-20:
Thank you @MichaelGrupp @7zhoulei . Looks like the most elegant way to solve this is to use datasets with extrinsic provided or to calibrate the extrinsic before using one's own data. Closing this issue now.

closed 2020-08-20 by zxl19