MichaelGrupp/evo #311: Question about multiple trajectories color

DarrenWong, 2020-08-06:
Dear Officer,

I am trying to plot 12 trajectories with different color but seems the color will repeat after 6 trajectories

`evo_traj tum 1.csv 2.csv 3.csv 4.csv 5.csv 6.csv 7.csv 8.csv 9.csv 10.csv 11.csv 12.csv --ref="1.csv" -p --plot_mode=xy
`
[image]

Tried to set the plot_seaborn_palette to some options but not working, can I have any other options to set 12 different colors?
`evo_config set plot_seaborn_palette "hls"
`

My config
```

 "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_pose_correspondences": false, 
    "plot_pose_correspondences_linestyle": "dotted", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "hls", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
```

DarrenWong, 2020-08-06:
Had a tmp fix way to hardcode the set_palette api, it would be ok for evo binary.   **n_colors=12** in  /usr/local/lib/python2.7/dist-packages/seaborn/rcmod.py

[image]

closed 2020-08-06 by DarrenWong

MichaelGrupp, 2020-08-06:
You can set `plot_multi_cmap` to a [colormap](https://matplotlib.org/3.1.1/tutorials/colors/colormaps.html).

```
plot_multi_cmap:
Color map for coloring plots from multiple data sources.
'none' will use the default color palette, see plot_seaborn_palette.
```

DarrenWong, 2020-08-07:
thx, it works  after setting `evo_config set plot_multi_cmap hsv`. Thanks a lot