MichaelGrupp/evo #315: Failed to load Python extension for LZ4 support. LZ4 compression will not be available.

autoperson, 2020-08-27:
In Ubuntu 18.04 ,
 Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/evo/entry_points.py", line 98, in launch
    main_module.run(args)
  File "/usr/local/lib/python3.6/dist-packages/evo/main_traj.py", line 297, in run
    trajectories, ref_traj = load_trajectories(args)
  File "/usr/local/lib/python3.6/dist-packages/evo/main_traj.py", line 229, in load_trajectories
    bag, topic)
  File "/usr/local/lib/python3.6/dist-packages/evo/tools/file_interface.py", line 249, in read_bag_trajectory
    from evo.tools import tf_cache
  File "/usr/local/lib/python3.6/dist-packages/evo/tools/tf_cache.py", line 27, in <module>
    import tf2_py
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

MichaelGrupp, 2020-08-27:
```
  File "/usr/local/lib/python3.6/dist-packages/evo/tools/tf_cache.py", line 27, in <module>
    import tf2_py
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
```

From the data provided I can only see that you have evo installed for Python 3 and ROS Melodic on Ubuntu 18 uses Python 2, so this can't work. This is an expected error. You need to install evo for Python 2 in this case.

closed 2020-08-27 by MichaelGrupp

jingshaojing, 2021-01-21:
I have install evo for python2 ,when i run 
`evo_ape tum data.tum vio.txt -r trans_part --save_results result/ape.zip`
the errors occur:

`[ERROR] Unhandled error in evo.main_ape
Traceback (most recent call last):
  File "/home/user/.local/lib/python2.7/site-packages/evo/entry_points.py", line 98, in launch
    main_module.run(args)
  File "/home/user/.local/lib/python2.7/site-packages/evo/main_ape.py", line 267, in run
    confirm_overwrite=not args.no_warnings)
  File "/home/user/.local/lib/python2.7/site-packages/evo/tools/file_interface.py", line 341, in save_res_file
    write_tum_trajectory_file(buffer, traj)
  File "/home/user/.local/lib/python2.7/site-packages/evo/tools/file_interface.py", line 134, in write_tum_trajectory_file
    np.savetxt(file_path, mat, delimiter=" ")
  File "/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 1258, in savetxt
    % (str(X.dtype), format))
TypeError: Mismatch between array dtype ('float64') and format specifier ('%.18e %.18e %.18e %.18e %.18e %.18e %.18e %.18e')

[ERROR] evo module evo.main_ape crashed - no logfile written (disabled)`

MichaelGrupp, 2021-01-21:
@jingshaojing what's your numpy version?