MichaelGrupp/evo #320: Rotate/zoom xyz plot

lpdon, 2020-09-12:
**Description:**
I am used to mplot3d and would expect it would be possible to zoom in/out and rotate the plot when in xyz mode, but it is not possible i.e the plot is always locked in the default configuration. It does work fine when only two coordinates are used to plot the trajectory or in the xyz_view and rpy_view tabs. I am not even sure this is a bug either. 

**Command:**
```
# replace this line with the command(s) you used
 evo_traj tum data.txt -p
```

**Console output:**
```
name:   data
infos:  27 poses, 1.539m path length, 0.520s duration

```

**Additional files:**
Please attach all the files needed to reproduce the error.
[data.txt](https://github.com/MichaelGrupp/evo/files/5212710/data.txt)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.11.0
* Python version shown by `evo pkg --pyversion`: 3.6.9
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.1,
    "plot_backend": "Qt5Agg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_unknown_cell_value": 205,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2020-09-14:
The zoom button does not work in mplot3d, you have to do it like this:

> One can rotate the 3D scene by simply clicking-and-dragging the scene. Zooming is done by right-clicking the scene and dragging the mouse up and down. Note that one does not use the zoom button like one would use for regular 2D plots.

https://matplotlib.org/mpl_toolkits/mplot3d/index.html

lpdon, 2020-09-18:
This is actually the procedure I tried. With mplot3d it works, but this is not the case for evo. I would have to try with another setup to see if it is reproducible. For my current needs the default camera pose is nevertheless more than fine. Btw, evo is a very nice package. Thanks!

closed 2020-09-18 by lpdon