MichaelGrupp/evo #321: Relative Rotational and Translation Errors for KITTI dataset

shubhendu1907, 2020-09-26:
Thank you @MichaelGrupp for this amazing analysis tool.

How can I calculate the relative rotational and translation errors for KITTI dataset? I tried using main_rpe.py but was not able to replicate the results of ORB-SLAM2 for KITTI ground-truth. The Absolute Pose Error (main_ape.py) did work fine. Just confused as to what parameters to use. Here is the one that I tried: 

python main_rpe.py kitti ~/Ground-Truth-KITTI/GT-10.txt ~/ORB-SLAM2/ORB-SLAM2-10.txt -va -d 100 -u m -r angle_deg --plot --plot_mode=xz

Any help is appreciated. Thank you.

MichaelGrupp, 2020-09-28:
Hi, a similar question was asked here: https://github.com/MichaelGrupp/evo/issues/144

My [answer](https://github.com/MichaelGrupp/evo/issues/144#issuecomment-464330736) there also applies to your question.

shubhendu1907, 2020-09-28:
Thank you for the link. And once again thanks for the amazing tool. 
I will close this thread.

closed 2020-09-28 by shubhendu1907