MichaelGrupp/evo #322: Plot RPE against time

dwisth, 2020-09-29:
Hello, great work on this package!

Just a quick question, is there any way to plot RPE against **time** instead of **index**? It would be a bit more intuitive to analyse.

This is what I get when I run the script:
[image]

Thanks for your help!

MichaelGrupp, 2020-09-30:
Index is shown if you are using input without timestamps. I assume you use a file format without timestamps?

mcamurri, 2020-10-05:
I think the problem is related with the `--all_pairs` option. 

Given two files with the exact same timestamps but different values, called `gt.csv` and `test.csv`, the following command would produce plots against time with a delta of `10 m`:
```
evo_rpe tum gt.csv test.csv --plot -u m -d 10
```
while the following one would produce it against index:
```
evo_rpe tum gt.csv test.csv --plot -u m  -d 10 --all_pairs
```

As discussed in a [previous issue](https://github.com/MichaelGrupp/evo/issues/301), if the trajectory is short compared to the delta, `--all_pairs` should be used to avoid sparse plots, but plotting the trajectory error is possible (up to the last delta).

For the same reason, it should also be possible to plot the RPE against time, using the timestamp at the start of the delta as a convention.

Am I right, @MichaelGrupp ?

MichaelGrupp, 2020-10-07:
Good point, @mcamurri, I opened a PR to address your point.

(my first comment above is still valid with the patch of course, index is shown if there are no timestamps)

closed 2020-10-08 by MichaelGrupp