MichaelGrupp/evo #329: found no matching timestamps between reference and ORBSLAM3.txt

gp1998-star, 2020-10-19:
**Description:**
Thank you for your outstanding work！
When I compare the trajectory of orbslam3 and groundtruth, there is a time stamp error.
I would appreciate it if you could give me some pointers.

**Command:**
```
evo_traj tum ORBSLAM3.txt --ref=groundtruth.txt -p -as
```

**Console output:**
```

[ERROR] found no matching timestamps between reference and ORBSLAM3.txt with max. time diff 0.01 (s) and time offset 0.0 (s)

```

**Additional files:**
Please attach all the files needed to reproduce the error.
[ORBSLAM3.txt](https://github.com/MichaelGrupp/evo/files/5400587/ORBSLAM3.txt)
[groundtruth.txt](https://github.com/MichaelGrupp/evo/files/5400591/groundtruth.txt)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.7.2
* Python version shown by `evo pkg --pyversion`: 2.7.17
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 18.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

MichaelGrupp, 2020-10-19:
ORBSLAM3.txt has the timestamps not in seconds as it looks like. The expected time unit is seconds. You can use [a script](https://github.com/MichaelGrupp/evo/blob/master/contrib/multiply_timestamps.py) to fix it, but it looks like a bug in the SLAM system if it writes the file like that.

```
evo_traj tum ORBSLAM3.txt --ref groundtruth.txt --verbose 
```

```
--------------------------------------------------------------------------------
Loaded 3680 stamps and poses from: ORBSLAM3.txt
Loaded 36382 stamps and poses from: groundtruth.txt
--------------------------------------------------------------------------------
name:	ORBSLAM3
infos:
	duration (s)	183949999872.0
	nr. of poses	3680
	path length (m)	68.8404436167
	pos_end (m)	[-0.10982274  0.24880588  0.11452151]
	pos_start (m)	[-3.8739000e-05  2.4535727e-02  1.0230887e-02]
	t_end (s)	1.4036367638135555e+18
	t_start (s)	1.4036365798635556e+18
--------------------------------------------------------------------------------
name:	groundtruth
infos:
	duration (s)	181.90499997138977
	nr. of poses	36382
	path length (m)	80.6262118473
	pos_end (m)	[ 4.572531 -1.942023  0.61962 ]
	pos_start (m)	[ 4.688319 -1.786938  0.783338]
	t_end (s)	1403636762.7435555
	t_start (s)	1403636580.8385556
```

gp1998-star, 2020-10-19:
> ORBSLAM3.txt has the timestamps not in seconds as it looks like. The expected time unit is seconds. You can use [a [...]
Thank you so much! 
Use multiply_timestamps.py to align the timestamp of ORBSLAM3 with groundtruth to solve the problem.

gp1998-star, 2020-10-19:
solved

closed 2020-10-19 by gp1998-star

MichaelGrupp, 2020-10-19:
Great :smile: If you experienced this with the official ORB SLAM 3 code, I would recommend to make a bug issue there. This output looks wrong to me.

gp1998-star, 2020-10-20:
> Great  If you experienced this with the official ORB SLAM 3 code, I would recommend to make a bug issue there. This output looks wrong to me.

sorry,I am not very familiar with github, how can I do it?

MichaelGrupp, 2020-10-21:
I mean if you are using the official ORB-SLAM 3 without changes, it would make sense to report them that they write timestamps wrong. But I don't know if this is the case.

https://github.com/UZ-SLAMLab/ORB_SLAM3 -> issues

gp1998-star, 2020-10-21:
I understand

xhding1997, 2021-05-07:
> ORBSLAM3.txt has the timestamps not in seconds as it looks like. The expected time unit is seconds. You can use [a script](https://github.com/MichaelGrupp/evo/blob/master/contrib/multiply_timestamps.py) to fix it, but it looks like a bug in the SLAM system if it writes the file like that.

Hellow, I dont know how to use this script. 
Can you tell me more about how to use it ?
Such as a command like `evo_traj tum f_dataset-MH01_monoi.txt --ref f_dataset-MH01_monoi.txt -p --plot_mode xyz`.
If I have two file, just like `mytraj.txt` and `groundtruth.tum`, and the txt file timestamp is different from the tum file.