MichaelGrupp/evo #330: Change Trajectory Plot Title

jackteversham, 2020-10-26:
Hi Michael

Is it possible to change/add a title to the trajectory plot? It currently doesn't have one.
How might I add if I've installed evo from source?

Thanks, Jack

MichaelGrupp, 2020-10-26:
The plots are made with matplotlib, so you can change them in the code like any other matplotlib axes / figure object.

Try adding `plt.title("test")` or `ax.set_title("test")` in this function:
https://github.com/MichaelGrupp/evo/blob/60b7927c0838240be87200c444d7dc2949eb44c6/evo/tools/plot.py#L310

https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.title.html
https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.axes.Axes.set_title.html

jackteversham, 2020-10-27:
Ah this worked - could have sworn I tried that and got an error before. Anyway - a customisable title might be a nice feature to add to the config settings so that people don't have to fiddle with source code.
Thanks Michael - great tool by the way.

closed 2020-10-27 by jackteversham