MichaelGrupp/evo #331: Seems like that the evo_ape treat the pose from my ground truth file as int instead of float.

Sologala, 2020-11-06:
**Description:**
Hi , recnetly , I have used the *evo* tools  to evaluate my alogrithm , and it is very nice and useful. But , I got some wrong results when I use it to  evaluate my own dataset which is sampled from an simulated 3D model. 

For the reason that this two follow files are too long , I just post a few part of them.
**ground truth file**
```sh
1604644894.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 1.000000
1604644894.000000 -0.331901 0.030874 0.000000 0.000000 0.000000 0.000000 1.000000
1604644894.000000 -0.663802 0.061749 0.000000 0.000000 0.000000 0.000000 1.000000
1604644894.000000 -0.995702 0.092623 0.000000 0.000000 0.000000 0.000000 1.000000
1604644894.000000 -1.327603 0.123497 0.000000 0.000000 0.000000 0.000000 1.000000
1604644894.000000 -1.659503 0.154371 0.000000 0.000000 0.000000 0.000000 1.000000
1604644894.000000 -1.992010 0.177842 0.000000 0.000000 0.000000 0.000000 1.000000
........
``` 
** this is the plot-image of  ground truth trajectory by evo_traj , the output trajectory is also correct** 
[image]

**output trajectory**
```sh
1604644894.000000 0.000000000 0.000000000 0.000000000 0.000000000 0.000000000 0.000000000 1.000000000
1604644894.000000 -0.324178934 0.043659735 0.031659655 0.000088743 -0.000102886 0.000013495 1.000000000
1604644894.000000 -0.636443317 0.071154483 0.013792973 0.000122277 -0.000339326 -0.000032323 0.999999940
1604644894.000000 -0.981475353 0.082766756 0.011885623 -0.000132466 -0.000180504 0.000125543 0.999999940
1604644894.000000 -1.304092526 0.134626746 0.018588986 0.000070129 -0.000325064 0.000075051 0.999999940
1604644894.000000 -1.635681748 0.189668849 0.033506855 0.000344531 -0.000244685 0.000113682 0.999999881
1604644894.000000 -1.983590603 0.204917401 0.045015968 0.000236127 -0.000132661 0.000005387 0.999999940
```
**commad line code that I used **
```sh
evo_ape tum ~/SLAM/Datasets/data/trajectory.txt ./forward.txt -p  -va
```

The results generated by evo are like this.

[image]

why the reference trajectory (dash-line) are composed of segments of polylines  instead of cruve  in evo_ape output image. 

Is there come args error in my command line code ?

Sologala, 2020-11-06:
and here are result without align

**command line code **

```sh
evo_ape tum ~/SLAM/Datasets/data/trajectory.txt ./forward.txt -p  
```
[image]

At the top of graph, It echo **Error mapped onto trajectory**

MichaelGrupp, 2020-11-08:
Look at the timestamps: you have multiple times the same timestamp, but different poses. This doesn't make sense.

Sologala, 2020-11-09:
> Look at the timestamps: you have multiple times the same timestamp 1604644894.000000, but different poses. This doesn't make sense.

Got it~ 
By reassigned the timestemps, It work very well! THX :D.

closed 2020-11-09 by Sologala