MichaelGrupp/evo #348: The trajectory alignment problem running the tum data room1

YitongD, 2020-11-11:
Hi,there is a problem when using evo to compare two tum trajectory.

[image]

But it’s normal to display the trajectory separately.

MichaelGrupp, 2020-11-11:
Just based on this picture, I don't understand what you think is wrong here.

YitongD, 2020-11-20:
Thanks，I have solved that problem which occured because of the timestamp.
And  I would like to ask when using the command "evo_traj" to evaluate, does the displayed  "Rotation of alignment" mean to directly transform the coordinates?
as:
Rotation of alignment: A[3,3]
Translation of alignment: B[1,3]
Two tracks: R(x,y,z) P(x,y,z)
R(x)=P(x)*A[0,0]+P(y)*A[0,1]+P(z)*A[0,2]+B[0,0]
R(y)=P(x)*A[1,0]+P(y)*A[1,1]+P(z)*A[1,2]+B[0,1]
R(z)=P(x)*A[2,0]+P(y)*A[2,1]+P(z)*A[2,2]+B[0,2]

THANK YOU

XuZeDong-xzd, 2020-12-16:
you can use parameter that "-a --correct_scale" solve this problem

MichaelGrupp, 2021-01-08:
That is the SO(3) rotation matrix that is used to transform each pose of the trajectory. See the code for details, it's part of the SE(3) matrix `t` that is used here: https://github.com/MichaelGrupp/evo/blob/a0b95e5fb4503c7165a687a3b074e94658fc4fac/evo/core/trajectory.py#L167

closed 2021-01-08 by MichaelGrupp