MichaelGrupp/evo #349: save_as_bag error if ros topic name have some prefix and same suffix

CrazyPopLin, 2020-11-17:
**Description:**

When I use the save_as_bag option in evo_traj to save rosbag, the input is also a rosbag. If the ground truth ros topic name is : **/nav/processed/odometry**; the estimated ros topic is: **/test/odometry**

Then I will got new saved rosbag, but the rostopics are all **odometry**.

If I changed the names using rosbag_editor before feed to evo_traj, it works fine. 

**Command:**
```
evo_traj bag firstCircle_twoOdom.bag /test/odometry --ref /nav/processed/odometry --full_check -p --plot_mode=xy --transform_left transf_stable_yaw.json --save_as_bag
```

**Console output:**
```

--------------------------------------------------------------------------------
name:	/test/odometry
infos:
	duration (s)	942.6128523349762
	nr. of poses	94242
	path length (m)	5187.9907899383925
	pos_end (m)	[-131.49911389  126.17903102    2.65406585]
	pos_start (m)	[-76.82377415  19.93870939  -1.94169016]
	t_end (s)	1599153462.9778523
	t_start (s)	1599152520.365
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	ok
	timestamps	wrong, not ascending or duplicates
stats:
	error - bad timestamps: 1599152520.410311 & 1599152520.395
--------------------------------------------------------------------------------
name:	/nav/processed/odometry
infos:
	duration (s)	942.5360000133514
	nr. of poses	47128
	path length (m)	2625.517289926155
	pos_end (m)	[-108.6835099   127.75241721   -1.96689975]
	pos_start (m)	[-76.82377415  19.93870939  -1.94169016]
	t_end (s)	1599153462.901
	t_start (s)	1599152520.365
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	ok
	timestamps	wrong, not ascending or duplicates
stats:
	error - bad timestamps: 1599152824.787624 & 1599152824.74
--------------------------------------------------------------------------------
Saving trajectories to 2020-11-17-12-17-03.bag...
Saved geometry_msgs/PoseStamped topic: odometry
Saved geometry_msgs/PoseStamped topic: odometry
QFileSystemWatcher::removePaths: list is empty
QFileSystemWatcher::removePaths: list is empty

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version: v1.12.0
* Python version: 3.8.5
* operating system: Ubuntu 20.04, ROS Noetic
* did you change the source code?: no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_unknown_cell_value": 205,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2020-11-18:
This looks like a bug. Thanks for reporting, I will have a look how to fix it when I have time.

MichaelGrupp, 2020-12-01:
Fixed in the newest version.

commit a0b95e5f (2020-12-01): Bump release version (closes #349 and #351).
closed 2020-12-01 by MichaelGrupp