MichaelGrupp/evo #353: What's the difference between `evo_ape` and TUM RGB-D's `evaluate_ate.py`

KaninchenM, 2020-12-03:
Are there any differences between`evo_ape` and [TUM RGB-D's tool- `evaluate_ate.py`](https://vision.in.tum.de/data/datasets/rgbd-dataset/tools)? Any differences in implementation details, such as the method of aligning timestamp, or the default parameters/threshold of calculation?

MichaelGrupp, 2020-12-03:
It's a completely different implementation.
I recommend to study the source code to understand the differences in detail.
`evaluate_ate.py` is one script, while `evo_ape` is built from a more modular basis.
Some details of evo metrics are also explained here: https://github.com/MichaelGrupp/evo/blob/master/notebooks/metrics.py_API_Documentation.ipynb

> method of aligning timestamp

evo is faster, but the matched timestamps are similar.

> default parameters

```
python evaluate_ate.py fr2_desk_groundtruth.txt fr2_desk_ORB.txt
```

is similar to

```
evo_ape tum fr2_desk_groundtruth.txt fr2_desk_ORB.txt --align
```

evo doesn't align by default.

KaninchenM, 2020-12-03:
> It's a completely different implementation.
> I recommend to study the source code to understand the differences in detail.
> `evaluate_ate.py` is one script, while `evo_ape` is built from a more [...]

I've tested on Euroc MH01 sequence. The result is same. Thank you very much!

closed 2020-12-03 by KaninchenM