MichaelGrupp/evo #354: Alignment flag can't be negated

YoshuaNava, 2020-12-03:
**Is your feature request related to a problem? Please describe.**
I have some bash scripts that I used to interact with evo. They feed the parameters to the tool, which can be provided manually or via ROS launchfiles, for example.

The align positional argument is giving me some issues, because I don't always want to specify Umeyama alignment. It only has two options, none of which negates the argument. Thus, if I want to give the user the option to run evo with or without alignment, I have to write extra bash code with one minor change; whether the alignment argument is provided or not.

**Describe the solution you'd like**
Have a third value for alignment, and be able to provide its value as `align=false`, for example.

**Describe alternatives you've considered**
Only use the Python API, which I already do, but for different purposes.

**Additional context**
Add any other context or screenshots about the feature request here.

MichaelGrupp, 2021-01-08:
This is how [argparse](https://docs.python.org/3/library/argparse.html) handles boolean flags, there's not really good way to circumvent this (except for changing the whole code to a completely different parser library that handles "true/false").

> Have a third value for alignment, and be able to provide its value as align=false, for example.
Not a fan of this, it would mean we need to this for every boolean flag.

I recommend to use JSON configuration files for automated scripts as described here: https://github.com/MichaelGrupp/evo/wiki/Configuration#custom-configuration-files
Users can then pass the config file path they want to use in your scenario.

YoshuaNava, 2021-01-18:
@MichaelGrupp Dear Michael, thank you very much. The JSON file config works great. Sorry for not checking that in the doc before opening the issue.

I consider this issue resolved and I'll close it.

closed 2021-01-20 by MichaelGrupp