MichaelGrupp/evo #355: evaluate on tum_mono datasets for dso

Mengyue1996, 2020-12-06:
**Description:**
It's look have some wrong question on tum_mono when I use the evo_traj to plot groundtruthSync.tst

**Command:**
```
# replace this line with the command(s) you used
```
evo_traj tum groundtruthSync.txt -p --plot_mode=xy
**Console output:**
```

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

MichaelGrupp, 2020-12-06:
??

Mengyue1996, 2020-12-09:
yeah ,thinks your attention, when I run dso_datasets in kitti sequence07,  find the result.txt have some question that
`evo_traj tum dso_result.txt -p`
[image]
[image]
after I use the code to translation result.txt to result.`evo_traj kitti dso_result.kitti -p`
[image]
but I want to contrast different result.txt by dso and ldso like evo_tum so that can be solve?

MichaelGrupp, 2021-01-09:
I really can't help you based on your vague description. But your plots show that timestamps are probably messed up, so you need to fix your data first.

closed 2021-01-09 by MichaelGrupp