MichaelGrupp/evo #356: Output translation and rotation computed by Umeyama as one of the APE/RPE metrics

YoshuaNava, 2020-12-07:
**Is your feature request related to a problem? Please describe.**
PR #108 introduces the possibility of extracting the transformation computed by Umeyama to align two trajectories.

However, the main_ape and main_rpe files, which serve as entry points for computing APE and RPE, do not allow you to extract these values, they are discarded by default.

**Describe the solution you'd like**
I would suggest that the translation and rotation computed by Umeyama is added to the metrics returned by main_ape and main_rpe.

**Describe alternatives you've considered**
None at the moment.

**Additional context**
None at the moment, but if needed I can provide further info about the use case, or anything else.

MichaelGrupp, 2021-01-12:
> I would suggest that the translation and rotation computed by Umeyama is added to the metrics returned by main_ape and main_rpe.

We can save it as additional metadata. But it is not part of the computed values of the metric IMO. The purpose of doing an alignment before computing the metric is to minimize the effect of misalignment or scale error on the result. In order to have a metric that incorporates these errors I would run it without alignment.

YoshuaNava, 2021-01-18:
> The purpose of doing an alignment before computing the metric is to minimize the effect of misalignment or scale error on the result. In order to have a metric that incorporates these errors I would run it without alignment.

I agree with you, and I personally run evaluations without Umeyama, only origin alignment. 

Nonetheless, I think Umeyama is useful when testing odometry methods in particular (wheel odometry, visual-inertial, LIDAR-based), to identify their absolute drift.

I remember the case of a colleague who used evo for comparing trajectories between a VIO system and a ground truth who was unable to find why such an estimate was creating issues when used as a reference by a robot prototype. The actual reason was that the IMU frame in one of the robots was 90 degree flipped with respect to gravity, and Umeyama was compensating for this orientation offset, and both RPE/APE were giving reasonable values, when the estimate was not (so) reasonable.
In this case, being able to look at the Umeyama transform would have made the developer aware of the strong orientation offset of the whole trajectory due to the flipped IMU frame.

MichaelGrupp, 2021-01-20:
As I said, I'm happy to include it as metadata in the result so the transformation matrix is accessible. This is only a small change in the code, see: https://github.com/MichaelGrupp/evo/pull/364

---

> Umeyama transform would have made the developer aware of the strong orientation offset of the whole trajectory due to the flipped IMU frame.

I don't think this is a reliable way to detect flipped frames. Consider the following dummy example of two trajectories that have identical positions, but flipped orientations.

[flipped.txt](https://github.com/MichaelGrupp/evo/files/5843597/flipped.txt)
[original.txt](https://github.com/MichaelGrupp/evo/files/5843598/original.txt)

[image]

In this case Umeyama would output an identity transformation, since it only takes the positions into account, and nothing would look suspicious with your approach.

The mismatch would be visible with `evo_ape tum original.tum flipped.tum --align --pose_relation angle_rad -v` though.

PR #364 (2021-01-20): Save alignment transformation in evo_{ape, rpe} result.
Saved as a Sim(3) matrix to accomodate all possible scenarios.

Included in the .zip file of `--save_results` and loadable via
```python
result = file_interface.load_res_file("test.zip")
t = result.np_arrays["alignment_transformation_sim3"]
```

Closes #356.
commit d96024d3 (2021-01-20): Save alignment transformation in evo_{ape, rpe} result. (#364)

Saved as a Sim(3) matrix to accomodate all possible scenarios.

Included in the .zip file of `--save_results` and loadable via
```python
result = file_interface.load_res_file("test.zip")
t = result.np_arrays["alignment_transformation_sim3"]
```

Closes #356.
closed 2021-01-20 by MichaelGrupp