MichaelGrupp/evo #357: pip install error: Syntax error in setup.py

zinuok, 2020-12-19:
**Description:**
I'm trying to install 'evo' using pip install from source, as I need to modify some script for my own use. However, I can't install 'evo'... ( I succeed to install evo from source a few months ago, but I fail this time)

**Command:** pip install --editable . --upgrade --no-binary evo
```
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///home/zinuok/evo
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/zinuok/evo/setup.py'"'"'; __file__='"'"'/home/zinuok/evo/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-aGyUFs
         cwd: /home/zinuok/evo/
    Complete output (6 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/zinuok/evo/setup.py", line 24
        print("Error:", e.output, file=sys.stderr)
                                      ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

```
**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: - (evo can't be installed yet)
* Python version shown by `evo pkg --pyversion`: pip 20.3.3 from /home/zinuok/.local/lib/python2.7/site-packages/pip (python 2.7)
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* did you change the source code? (yes / no): -
* output of `evo_config show --brief --no_color`: -
```

zinuok, 2020-12-19:
since python 2.7 is supported lower or equal than v.1.12.0, I cloned that version and this issue has resolved.

closed 2020-12-19 by zinuok

cheesekuntly, 2020-12-20:
@zinuok  Hi,I have the same problem as you, can you provide me with that version you has been resolved.