MichaelGrupp/evo #360: traj_colormap function breaks when using multiple figures

Usaali, 2020-12-28:
**Description:**
This is both an error report and a possible solution. I wasnt sure where to put it. I can create a pull request for this, if you want that.
I am using the traj_colormap function to plot the APE and RPE of trajectories and am writing a python script for that. Using that i realized, that the traj_colormap function breaks, if i use multiple figures. I have multiple colormaps in one figure and that works fine. but when i add another figure, the Trajectory plots stay on the original figure, but the colormap legend moves to the new figure. This renders the whole plot useless.
The reason for this is, that `matplotlib.pyplot.gcf()` is used to get the current figure for the colorbar. This breaks, if the plot is not done immediately after creating the figure and other figures are created in the meantime.
A fix i found for this is to give the traj_colormap function another argument and pass the current figure, instead of using gca()

**Command:**
Code to reproduce the behaviour:
```python
def plotApe(align, correct_scale):
    p = PlotCollection("Absolute Pose Error")
    figures={}
    #tum trajectory file paths
    truthFile =  "referenceFilePath"
    estFile =  "estimationFilePath"
    #save figures in a dictionary
    figures["3d_plot"] = plt.figure()
    figures["error_plot"] = plt.figure()
    #read files
    tTraj = file_interface.read_tum_trajectory_file(truthFile)
    eTraj = file_interface.read_tum_trajectory_file(estFile)
    tTraj, eTraj = sync.associate_trajectories(tTraj,eTraj)
    #prepare axis for trajectory plot
    ax = plot.prepare_axis(figures["3d_plot"],PlotMode.xyz)
    #calculate ape
    result_ape = main_ape.ape(tTraj, eTraj, est_name=(estFile.split("/")[-1]), pose_relation=PoseRelation.full_transformation, align=align, correct_scale=correct_scale)
    #plot reference
    plot.traj(ax, PlotMode.xyz, tTraj, style="--", alpha=0.5, label=(estFile.split("/")[-1]))
    #plot colormap
    plot.traj_colormap(ax, result_ape.trajectories[(estFile.split("/")[-1])], result_ape.np_arrays["error_array"], PlotMode.xyz, min_map=result_ape.stats["min"], max_map=result_ape.stats["max"], title=(result_ape.info["est_name"]+"\n"+result_ape.info["title"]))
    #add figures to PlotCollection
    p.add_figure("colormap",figures["3d_plot"])
    p.add_figure("empty fig",figures["error_plot"])

    p.show()
```
[image]
[image]

If the second figure is not generated, everything works fine. but otherwise, the colored bar gets plotted on the second figure instead of the first one

**Possible fix**
I changed the traj_colormap function in plot.py and added a parameter called fig, which defaults to None. Then if fig is None, it gets created with plt.gca(). Otherwise the given figure is used. This should keep backwardscompatibility and not break on older versions
```python
def traj_colormap(ax, traj, array, plot_mode, min_map, max_map, title="", fig = None):
    """
    color map a path/trajectory in xyz coordinates according to
    an array of values
    :param ax: plot axis
    :param traj: trajectory.PosePath3D or trajectory.PoseTrajectory3D object
    :param array: Nx1 array of values used for color mapping
    :param plot_mode: PlotMode
    :param min_map: lower bound value for color mapping
    :param max_map: upper bound value for color mapping
    :param title: plot title
    """
    pos = traj.positions_xyz
    norm = mpl.colors.Normalize(vmin=min_map, vmax=max_map, clip=True)
    mapper = cm.ScalarMappable(
        norm=norm,
        cmap=SETTINGS.plot_trajectory_cmap)  # cm.*_r is reversed cmap
    mapper.set_array(array)
    colors = [mapper.to_rgba(a) for a in array]
    line_collection = colored_line_collection(pos, colors, plot_mode)
    ax.add_collection(line_collection)
    ax.autoscale_view(True, True, True)
    if plot_mode == PlotMode.xyz:
        ax.set_zlim(
            np.amin(traj.positions_xyz[:, 2]),
            np.amax(traj.positions_xyz[:, 2]))
        if SETTINGS.plot_xyz_realistic:
            set_aspect_equal_3d(ax)
    if(fig == None):
        fig = plt.gcf()
    cbar = fig.colorbar(
        mapper, ticks=[min_map, (max_map - (max_map - min_map) / 2), max_map])
    cbar.ax.set_yticklabels([
        "{0:0.3f}".format(min_map),
        "{0:0.3f}".format(max_map - (max_map - min_map) / 2),
        "{0:0.3f}".format(max_map)
    ])
    if title:
        ax.legend(frameon=True)
        ax.set_title(title)
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.12.0
* Python version shown by `evo pkg --pyversion`: 2.7.17
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* did you change the source code? (yes / no): no (only to add the fix afterwards)
* output of `evo_config show --brief --no_color`: 
```
    "console_logging_format": "%(message)s", 
    "euler_angle_sequence": "sxyz", 
    "global_logfile_enabled": false, 
    "plot_axis_marker_scale": 0.0, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_mode_default": "xyz", 
    "plot_multi_cmap": "none", 
    "plot_pose_correspondences": false, 
    "plot_pose_correspondences_linestyle": "dotted", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "whitegrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
        "max"
    ], 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_trajectory_linestyle": "-", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "ros_map_alpha_value": 1.0, 
    "ros_map_unknown_cell_value": 205, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true, 
    "tf_cache_lookup_frequency": 10, 
    "tf_cache_max_time": 10000.0
```
I wanted some feedback, before i open a pull request, as i have never done something before. 
Thank you very much for your time

MichaelGrupp, 2021-01-08:
Thanks for the detailed explanation. We can use your patch. Do you want to make a PR? FYI: I see you're using Ubuntu 18 with Python 2 - note that I stopped supporting Python 2 after v1.12.0, so this will only be "officially" fixed for Python 3 once it is merged to the master branch.

Usaali, 2021-01-08:
I cereated a Pull Request for it #361 but the pipelines are failing right now. I am trying to get it running on python3 but the download for PyQt5 is currently going on with 25kB/s and failing after some time. Should i close the PR for now and open a new one when i am done, or just leave it open until then?
Unfortunately i am forced to use Python 2 because of other software i am using for my thesis. I will just keep the fix locally then. Thanks for letting me know that the change will only affect Python3.

closed 2021-01-10 by MichaelGrupp