MichaelGrupp/evo #366: [ERROR] Unhandled error in evo.main_ape

jingshaojing, 2021-01-22:
**Description:**
when I want to run 

**Command:**
```
evo_ape tum data.tum vio.txt -p -va -r trans_part --save_results result/ape.zip
```
I get :
**Console output:**
```

#Saving results to result/ape.zip...
[ERROR] Unhandled error in evo.main_ape
Traceback (most recent call last):
  File "/home/user/.local/lib/python2.7/site-packages/evo/entry_points.py", line 98, in launch
    main_module.run(args)
  File "/home/user/.local/lib/python2.7/site-packages/evo/main_ape.py", line 267, in run
    confirm_overwrite=not args.no_warnings)
  File "/home/user/.local/lib/python2.7/site-packages/evo/tools/file_interface.py", line 341, in save_res_file
    write_tum_trajectory_file(buffer, traj)
  File "/home/user/.local/lib/python2.7/site-packages/evo/tools/file_interface.py", line 134, in write_tum_trajectory_file
    np.savetxt(file_path, mat, delimiter=" ")
  File "/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 1258, in savetxt
    % (str(X.dtype), format))
TypeError: Mismatch between array dtype ('float64') and format specifier ('%.18e %.18e %.18e %.18e %.18e %.18e %.18e %.18e')

```

**Additional files:**
Please attach all the files needed to reproduce the error.
[data.zip](https://github.com/MichaelGrupp/evo/files/5852885/data.zip)
(data.tum and vio.txt are include in this .zip file)

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

jingshaojing, 2021-01-22:
when I upgrade numpy with ` pip install numpy --upgrade --user`, the question about `evo_ape  (rpe)` has been solved ,but the question about `evo_res ` is not solved as follows:
run :`evo_res result/*.zip -p --save_table result/table.csv`
get 
`[ERROR] Values of 'est_name' must be unique - duplicates: vio.txt, vio.txt
Try using the --use_filenames option to use filenames for labeling instead.`

jingshaojing, 2021-01-22:
my new question is I can run `evo_res ape.zip -p` or `evo_res rpe.zip -p` without problem,
but when I used 
` user@lenovo:~/MSCKFVIO2_ws/result$ evo_res ./ape.zip rpe.zip -p  `or 
`buaa@lenovo:~/MSCKFVIO2_ws/result$ evo_res ape.zip rpe.zip -p `
it has problem :
`[ERROR] Values of 'est_name' must be unique - duplicates: vio.txt, vio.txt
Try using the --use_filenames option to use filenames for labeling instead`

MichaelGrupp, 2021-01-22:
> Try using the --use_filenames option

Did you try that?

jingshaojing, 2021-01-22:
> > Try using the --use_filenames option
> 
> Did you try that?t 

   I  run with `evo_res result/*.zip --use_filenames -p`, it make sense

closed 2021-01-22 by MichaelGrupp