MichaelGrupp/evo #379: ERROR file format with gt TUM files

GabHoo, 2021-02-14:
I know this sound so weird, I am definitely missing something.. I am getting the "[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)" when trying the evo_traj with TUM gt file. 
The file is straight out of the dataset folder and untuched. The format should be the one "tum" option is based on right?

command:
evo_traj tum /dataset_room1/mav0/mocap0/data.txt 
[data.txt](https://github.com/MichaelGrupp/evo/files/5976867/data.txt)
 
evo version: v1.13.3
py version: 3.8.5
Ubuntu 20.04
code is unchanged

Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE
evo version: v1.13.3
py version: 3.8.5
Ubuntu 20.04
code is unchanged

[data.txt](https://github.com/MichaelGrupp/evo/files/5976870/data.txt)

```

chunibyo-wly, 2021-02-14:
Hello, after checking your file, I noticed that your format is not follow tum format maybe.
refer to this [link](https://github.com/MichaelGrupp/evo/wiki/Formats#tum---tum-rgb-d-dataset-trajectory-format).
delete comma and change the order of sequence will work.

You can simply execute this to meet format requirements.
`awk -F ',' '{print $1" "$6" "$7" "$8" "$2" "$3" "$4" "$5}' data.txt > out.txt`

[image]

GabHoo, 2021-02-14:
Right! Thank you so much for your fast response and kindness.

The file evo takes as standard TUM trajectory is the ground trouth from the RGB-D dataset and not the VISUAL INERTIAL dataset.
It's a matter of commas and order of colums tho, could have seen that by myself.

But I was wandering: shouldn't it be like: 

$ awk -F ',' '{print $1" "$2" "$3" "$4" "$6" "$7" "$8" "$5}' data.txt > out.txt

Because the format of the gt file is actually time tx ty tz qw qx qy qz , that means we have to swap the qw column at the end of the row to have it tx ty tz qx qy qz qw don't we?

Thank you very much!

chunibyo-wly, 2021-02-14:
yes, my mistake, sorry for that XD.

closed 2021-02-14 by GabHoo

MichaelGrupp, 2021-02-14:
It's confusing because the TUM VI dataset actually follows the EuRoC dataset format, not the TUM RGB-D format. You can see it in the download tables here, it says "Euroc / DSO": https://vision.in.tum.de/data/datasets/visual-inertial-dataset

I made a small update to support also loading the TUM VI groundtruth with `euroc` mode.
https://github.com/MichaelGrupp/evo/commit/8b817b8e3cc25272814ed086a13077b24ca7f4a0

After you updated, you can use the TUM VI dataset as described in the euroc format documentation :wink: https://github.com/MichaelGrupp/evo/wiki/Formats#euroc---euroc-mav-dataset-format-also-used-in-tum-vi-dataset

MichaelGrupp, 2021-02-14:
> `awk -F ',' '{print $1" "$2" "$3" "$4" "$6" "$7" "$8" "$5}' data.txt > out.txt`

The euroc format uses nanoseconds for the timestamps, and evo expects seconds for TUM. If you convert it with the `--save_as_tum` as written in my comment above this will be converted.