MichaelGrupp/evo #381: Few Simple Introductive questions

GabHoo, 2021-02-14:
Hi There, first of all thank you for the amazing work which, as it looks like from the community, was definitely needed.
I am just new to SLAM and trajectory evaluation and I am still reading docs and wikis but I do have few basic questions I can't wrap my head around (I hope they will be helpful to other beginners)

1) What is the difference between pose and trajectory? I saw both words being used for pretty much every kind of datas. It make sense to me to call pose the calculated points where every timestamp is associated with its position compared to the initial 0,0,0 point of the first timestamp available, while trajectory could be every timestamp is associated with its coordinates considering the previous timestamp as starting point. That would make a pose file made of incremental measurements for example. Is this right? I guess the output of a SLAM algorithm can be given in both ways but is this aspect take care of during evaluation? Moreover are this pose just cordinates ore trasformation? in case of trasformation do they refer from the frame to get back to origin or viceversa? 

2)I saw that in order to compare two or more trajectories you first need to associate the timestamps since the files you want to compare probably have different number of total poses at different times. Is this association made inside every evo function like rpe ape?

3)I also noticed that when calculating ape you first need to align trajectories one to another ( If I got it right there is no need to do that in the rpe since you are considering deltas). What does that really mean? Is this necessary also in evo_traj? I also saw there are a bunch of different methods for doing that...

4)Is that true that you consider only translations while comparing trajectorys? If so do you mean alignment is also done just for translation?

5)Since Kitti dataset_gt has not timestamp inside, I saw that a script has been made that actually associate it and returning a TUM format file. Does that actually convert the 3x3 rot matrix and translation to a quaternion + translation? Are there any loss of precision in such process?

Thanks in advanced for those who will find time and patience to answer.

MichaelGrupp, 2021-02-16:
> What is the difference between pose and trajectory?

* pose: position and orientation
* path: sequence of consecutive poses
* trajectory sequence of consecutive poses with timestamp per pose

> Is this association made inside every evo function like rpe ape?

Yes. For evo_traj it is done if needed, see https://github.com/MichaelGrupp/evo/wiki/evo_traj#temporal-alignment

> I also noticed that when calculating ape you first need to align trajectories one to another

This is optional and depends on what you want to measure.
https://github.com/MichaelGrupp/evo/wiki/Metrics#alignment
https://github.com/MichaelGrupp/evo/issues/268

> If I got it right there is no need to do that in the rpe since you are considering deltas

Alignment with scaling will have an effect on translational RPE.

> Is that true that you consider only translations while comparing trajectorys? If so do you mean alignment is also done just for translation?

You can change which property is compared with `--pose_relation`. See here for a similar question: https://github.com/MichaelGrupp/evo/issues/62

Umeyama alignment is pure point alignment, so no rotation is used there. If you do the origin alignment the origin frames are aligned also with respect to rotation.

> KITTI to TUM script

It converts the matrix to translation + quaternion as this is required for the TUM text format. There will always be some very small loss of precision. Should be hard to measure.

GabHoo, 2021-02-17:
Thak you for your time!

closed 2021-02-17 by GabHoo