MichaelGrupp/evo #390: Illegal instruction error

Callay15, 2021-03-17:
**Description:**
I cannot seem to run any of the commands with the  .txt files. I can run evo_traj --help and evo_traj kitti shows the help instructions but the moment a KITTI or TUM file goes in, it returns a Illegal instruction(core dumped). Even the demos didnt work.  Ive have tried running one of the matplotlib examples and had no issues.

**Command:**
```
# evo_traj kitti ~/ORB_SLAM2/FrameTrajectory_KITTI_Format.txt 

```

**Console output:**
```

# Illegal instruction (core dumped)

```

**Additional files:**
Please attach all the files needed to reproduce the error.
[image]

Please give also the following information:
* evo version number shown by `evo pkg --version`: 1.13.4
* Python version shown by `evo pkg --pyversion`: 3.6.9
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04 LTS
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

# {
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_unknown_cell_value": 205,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2021-03-17:
Illegal instruction would be the case if you run code compiled for a different CPU architecture. Since evo itself is pure Python, this rather comes from one of the binary dependencies like for numpy, which is based on C-code.

You seem to be using an Nvidia Jetson, so it looks like something is not properly installed for the ARM CPU of your board.
Could be this here: https://github.com/numpy/numpy/issues/18131, or something else. I can't help much more from my side.

Callay15, 2021-03-18:
Yes I am running on a Jetson Nano, it seems the ARM CPU was the issue here. Following this comment from you [link](https://github.com/numpy/numpy/issues/18131#issuecomment-762235284) solved the issue. Thank you very much!

closed 2021-03-18 by Callay15