MichaelGrupp/evo #391: evo_traj: rpy_view

jingshaojing, 2021-03-23:
when i run `evo_traj tum vio.txt --ref data.tum -p -va -s`, where vio.txt comes from s-msckf on eurov v1_01.bag, data.tum is respondding ground truth
the rpy_view seems pitch ,roll yaw angle have  a big difference as follows
[image]
espically pitch angle.
and  I use `evo_ape` for angle deg as follows:
`evo_ape tum data.tum vio.txt -r angle_deg
APE w.r.t. rotation angle in degrees (deg)
(not aligned)

       max	179.999505
      mean	168.936737
    median	173.688309
       min	136.357055
      rmse	169.318499
       sse	82336661.610298
       std	11.363664`
RMSE and STD are so large, So is this angle display normal? And whether this error is normal, feel the pitch angle is opposite

MichaelGrupp, 2021-04-14:
Are you sure you don't have rotated frames in your trajectory? (for example, this could be checked visually by plotting axis markers https://github.com/MichaelGrupp/evo/wiki/Plotting#coordinate-axis-markers)

In other words: is your groundtruth trajectory describing the same frame as your estimate? (extrinsic calibration)

jingshaojing, 2021-04-14:
so,if my coordinates are not consistent, should I  add some parameter about extrinsic calibration? sorry ,I have not  seen how to do?

MichaelGrupp, 2021-04-14:
You need to check you data, I can't help you with that. For example: if your trajectory describes the path of the IMU frame, but the groundtruth describes the path of the camera frame, you will get errors.

jingshaojing, 2021-04-14:
so it means I  can not compare attiude in different frames with evo?  I have to trans it to same frame?

MichaelGrupp, 2021-04-14:
You want to measure rotation error, so there should be no systematic rotation error in your trajectory. So yes, your trajectories must describe the same semantic path of course.

jingshaojing, 2021-04-14:
thanks for your reply

closed 2021-04-14 by MichaelGrupp