MichaelGrupp/evo #393: [ERROR] found no matching timestamps between xxx and xxx

heirenlop, 2021-04-05:
**Description:**
Thank you for your outstanding work！
When I compare the trajectory of lio-sam and kitti_05_gt, there is a time stamp error.
I would appreciate it if you could give me some pointers.
odometry_incremental.tum is the odometry.tum file of lio-sam.

**Command:**
```
evo_traj tum odometry_incremental.tum --ref=kitti_05_gt.txt -p --plot_mode xyz -a
```

**Console output:**
```

[ERROR] found no matching timestamps between reference and odometry_incremental.tum with max. time diff 0.01 (s) and time offset 0.0 (s)

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.12.0
* Python version shown by `evo pkg --pyversion`: 2.7.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 16.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
{
    "console_logging_format": "%(message)s", 
    "euler_angle_sequence": "sxyz", 
    "global_logfile_enabled": false, 
    "plot_axis_marker_scale": 0.0, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_mode_default": "xyz", 
    "plot_multi_cmap": "none", 
    "plot_pose_correspondences": false, 
    "plot_pose_correspondences_linestyle": "dotted", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
        "max"
    ], 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_trajectory_linestyle": "-", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "ros_map_alpha_value": 1.0, 
    "ros_map_unknown_cell_value": 205, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true, 
    "tf_cache_lookup_frequency": 10, 
    "tf_cache_max_time": 10000.0
}

```

# remove this line and paste your config HERE

```

heirenlop, 2021-04-06:
I found that the timestamps of my two files are not aligned. The information of the two files is as follows:
`name:	odometry_incremental
infos:
	duration (s)	287.3287880420685
	nr. of poses	1376
	path length (m)	2200.88142225
	pos_end (m)	[-73.35428619 355.67181396  -3.41995358]
	pos_start (m)	[0. 0. 0.]
	t_end (s)	1317355166.560833
	t_start (s)	1317354879.232045
--------------------------------------------------------------------------------
name:	kitti_05_gt
infos:
	duration (s)	287.537
	nr. of poses	2761
	path length (m)	2205.57576386
	pos_end (m)	[ -4.804541 -10.99719  370.2569  ]
	pos_start (m)	[1.110223e-16 0.000000e+00 2.220446e-16]
	t_end (s)	287.537
	t_start (s)	0.0
`

How can I solve this problem

heirenlop, 2021-04-07:
sorry, i have not noticed --t_offset

closed 2021-04-07 by heirenlop

kc944201, 2021-07-01:
@heirenlop could you please teach me how to align timestamps?
Because I use  --t_offset and --t_max_diff, the results are wrong.

misaya1218, 2021-12-06:
> 

evo_traj tum re_05.txt --ref=kitti_05_gt.txt -p --plot_mode xyz -a  --t_offset -1317354879.232044935