MichaelGrupp/evo #395: "evo_traj --transform_left" don't  work when with "--align"

TouchDeeper, 2021-04-11:
Hello,
I applied a transformation to trajectories by:
```
evo_traj tum --ref scale1/viwo/vio.csv scale1/vio/vio.csv --transform_left trans.json  --plot_mode xy --plot
```
and the transformation works.

Then I want to apply the transformation to the aligned trajectories by:
```
evo_traj tum --ref scale1/viwo/vio.csv scale1/vio/vio.csv --transform_left trans.json --align  --plot_mode xy --plot
```
But the transformation didn't work.

Could you please provide a quick code to apply a transformation to the aligned trajectories? Thanks a lot.

MichaelGrupp, 2021-04-11:
Yes, it makes more sense to do the transformation after the alignment step. I changed the order in the newest version.

closed 2021-04-11 by MichaelGrupp

TouchDeeper, 2021-04-12:
@MichaelGrupp 

It seems the newest version will not transform the reference trajectory. I think the reference trajectory should be transformed together.

MichaelGrupp, 2021-04-12:
The reference trajectory is not transformed, anything else would be confusing. If you want to change the trajectory that you use as a reference, I recommend to transform it separately and save it to a new file before using it with your other trajectories:

```
`evo_traj tum <your reference file> --transform_left ... --save_as_tum
```

TouchDeeper, 2021-04-12:
@MichaelGrupp
Thank you for your advice. I think it is a good way.

commit ffda6b4b (2021-04-11): Apply custom transformation after alignment (fixes #395). (#397)