MichaelGrupp/evo #398: How to create a multiple boxplot with multiple deltas for Relative Error?

martinakos, 2021-04-13:
I first discovered this other project [rpg_trajectory_evaluation](https://github.com/uzh-rpg/rpg_trajectory_evaluation) for VIO evaluation. With this project I could create this type of multiple boxplots with multiple deltas for Relative Error:
[image]

However, I find rpg_trajectory_evaluation too cumbersome to use. Then I discovered evo. I think I prefer to use evo as with evo I can create 3D trajectories, analyse them and plot them from within my code without having to write files. I see it's also faster. However, I haven't found a way with evo to make a multiple boxplot for relative errors as the one above which are what I've been using so far. The closest I've been able to create is a relative error diagram for a single distance delta like this:

```
$ evo_rpe tum stamped_groundtruth.txt stamped_traj_estimate.txt -p -u m -d 1.87 --all_pairs
RPE w.r.t. translation part (m)
for delta = 1.87 (m) using all pairs
(not aligned)

       max	0.339555
      mean	0.221546
    median	0.218166
       min	0.101457
      rmse	0.226667
       sse	27.487257
       std	0.047909
```
[image]

I suppose this diagram is the equivalent for the first boxplot (at distance 1.87m) above. However, I don't understand the time axis and the statistics I get are slightly different from the ones I get from  rpg_trajectory_evaluation at 1.87m distance, which are:
```
at 1.87m
trans:
  max: 0.30304691384060745
  mean: 0.2089894788068043
  median: 0.20990006167173142
  min: 0.10585213477621969
  num_samples: 564
  rmse: 0.21299743629797438
  std: 0.041125486228979054  
```
Any idea why I get different statistics for the same data?
And it is possible with evo to create a multiple boxplot with multiple deltas (in m)  for Relative Error as the one above?

MichaelGrupp, 2021-04-14:
### Multiple box-plots

This is possible with evo, the difference to rpg_trajectory_evaluation is that this involves two steps to be more flexible:

* computing the metrics and saving the results
* creating the result comparison

(similar to the example in the README).

So you need to run the different metrics as you wish and save the results:

```
evo_rpe tum groundtruth.txt estimate.txt -d 1.87 -u m --all_pairs --save_results results/1.87m
...
```

Once you have saved all your results, you can run:
```
cd results/
evo_res 1.87m 3.75m --plot --use_filenames --ignore_title
```
and select the box plot. You might want to change the "estimate" label of the axis to "delta" - you can do this in the GUI if you are using the Qt backend for matplotlib.

---

### Different results between evo and rpg_trajectory_evaluation

You have to take into account that you are comparing two entirely different implementations. Numerical inaccuraciess alone would produce different results.

Considering this, the values are not far off. If you want to know why exactly they are different, you need to dig into the code of both implementations and compare what is calculated in both cases. E.g. how the delta filtering or the metrics computation differ.

martinakos, 2021-04-14:
That's great!  I can see how this two stage approach can be more flexible. 
Thanks for sharing this tool!

closed 2021-04-26 by MichaelGrupp