MichaelGrupp/evo #401: Python 3 alignment issue

phil333, 2021-04-26:
**Description:**

Different trajectories no longer work with python 3. Both for align trajectory and align trajectory at origin

**Command:**
```
trajectory.align_trajectory_origin(traj_est, traj_ref)
```

**Console output:**
```
AttributeError: module 'evo.core.trajectory' has no attribute 'align_trajectory'

```

MichaelGrupp, 2021-04-27:
This was changed from a function to a class method: https://github.com/MichaelGrupp/evo/commit/e96d6a33415cd3a5fd25fefa4a6a69ee70e900e4

phil333, 2021-04-27:
thank you, that worked. Just as reference to other people, you have to modify as follows:
from
```
traj_est = trajectory.align_trajectory_origin(traj_est, traj_ref)
```
to
```
traj_est.align_origin(traj_ref)
```

closed 2021-04-27 by phil333