MichaelGrupp/evo #402: evo_rpe and evo_ape reads EuRoC data with TUM format loader

Cryptum169, 2021-05-03:
**Description:**
`evo_rpe` and `evo_ape` reads EuRoC format files with TUM trajectory reader

https://github.com/MichaelGrupp/evo/blob/ffda6b4b91a2c5739f0d39cd3491ed5f4011d499/evo/common_ape_rpe.py#L50-L53 

**Command:**
For simplicity, ground truth file were passed to both argument.
```
evo_rpe euroc MH01_gt.csv MH01_gt.csv
```

**Console output:**
```
[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)
```

**Additional files:**
Please attach all the files needed to reproduce the error.
[MH01_gt.csv](https://github.com/MichaelGrupp/evo/files/6416778/MH01_gt.csv) (First 20 lines of EuRoC MH01 sequence ground truth estimate)

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```
evo pkg --version: v1.13.5
evo pkg --pyversion: 3.8.5
Ubuntu 20.04
no. 
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "TkAgg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_unknown_cell_value": 205,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2021-05-04:
This is expected behavior because the euroc file format is only useful for the groundtruth.

See here for more details:
https://github.com/MichaelGrupp/evo/wiki/Formats#euroc---euroc-mav-dataset-format-also-used-in-tum-vi-dataset

> evo_rpe euroc MH01_gt.csv MH01_gt.csv

I don't believe that comparison of a groundtruth file with itself is a common use case that would justify a change.

Cryptum169, 2021-05-04:
I see, I jumped to conclusions when I saw a TUM error popped up when euroc option was given, my apologies.

closed 2021-05-04 by Cryptum169

MichaelGrupp, 2021-05-04:
No problem, this might be confusing the first time :slightly_smiling_face: