MichaelGrupp/evo #406: Why do some curves become straight lines when comparing trajectories?

vivianwang233, 2021-05-21:
**Description:**
Why do some curves turn into straight lines when comparing trajectories, as shown by the blue line in the figure. 
When looking at the two trajectories separately, the contours of the two places are similar.
[image]

The command I used:  evo_traj tum t1.txt --ref t2.txt -p -as

MichaelGrupp, 2021-05-21:
You get a line when there's a jump. In your command the trajectories are time synced and the blue one only shows the matching poses. This completely depends on your input data, so you need to investigate it. Maybe these markers help: https://github.com/MichaelGrupp/evo/wiki/Plotting#pose-correspondence-markers

vivianwang233, 2021-05-21:
Thank you. Is there any way to show all poses in the blue line？The number of poses of the two lines is not the same.  I read the tutorial you mentioned but couldn’t find the answer.

MichaelGrupp, 2021-05-26:
The number of poses change because the trajectory is synchronized with the reference timestamps and the poses with non-matching timestamps are removed. In `evo_traj` you will however always see the full reference trajectory.

So I think you should investigate why you don't have matching timestamps with the reference in your data.

closed 2021-06-04 by MichaelGrupp