MichaelGrupp/evo #408: rpe plot with different axis

lintong-zhang, 2021-06-15:
Hi. Thank you for the great tool. Just want to check that if there is feature that I can make rpt plot with individual axis rather than all axis. This way it would be easier to inspect if there is a scale error, such in the x direction. Is this feature already in the code or we can implement quickly?

MichaelGrupp, 2021-06-24:
RPE is not calculated per axis, so there's no easy way to get what you want without custom code.

dwisth, 2021-06-24:
I just had a quick look through the code. If we want to implement this feature does it make sense to just add some new `PoseRelation` [enums](https://github.com/MichaelGrupp/evo/blob/a28849dbf119a6743e0cd00d06c1962f5b2a6bbc/evo/core/metrics.py#L60) (e.g. `rotation_part_roll`, `rotation_part_pitch`, `rotation_part_yaw`) then add new calculations for these pose relations (e.g. [APE](https://github.com/MichaelGrupp/evo/blob/a28849dbf119a6743e0cd00d06c1962f5b2a6bbc/evo/core/metrics.py#L355), [RPE](https://github.com/MichaelGrupp/evo/blob/a28849dbf119a6743e0cd00d06c1962f5b2a6bbc/evo/core/metrics.py#L269))?

closed 2021-09-13 by MichaelGrupp

MichaelGrupp, 2021-09-13:
@dwisth Could be added, but it sounds like something to be implemented for a specific use case in a fork of a repo. Because xcept for that one plot tab in `evo_traj`, I don't want Euler angles in the algorithms - they are error prone and you need to choose one of the many possible axis conventions to interpret them.

dwisth, 2021-09-22:
@MichaelGrupp - okay that makes sense. If we end up making these changes we'll let you know how it works.

Thanks for your help :)