MichaelGrupp/evo #411: The trajectory plot mode doesn't show anything

kc944201, 2021-06-25:
**Description:**
Hi, when I use evo_traj command want to plotting the trajectory
I use --plot_mode=xy, it didn't any reactions, just show the information about topic 
But when I use -p command, it can work to show xyz axes plot

I don't know why the --plot_mode is failure, I just need X、Y axis plot
Thank you!
**Command:**
```
evo_traj bag lawn.bag /odom --plot_mode=xy
```

**Console output:**
```
name:	/odom
infos:	5634 poses, 36.836m path length, 307.879s duration
```

**Additional files:**
[image]

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.12.0
* Python version shown by `evo pkg --pyversion`: 2.7.17
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu mate 18.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s", 
    "euler_angle_sequence": "sxyz", 
    "global_logfile_enabled": false, 
    "plot_axis_marker_scale": 0.0, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_mode_default": "xyz", 
    "plot_multi_cmap": "none", 
    "plot_pose_correspondences": false, 
    "plot_pose_correspondences_linestyle": "dotted", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
        "max"
    ], 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_trajectory_linestyle": "-", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "ros_map_alpha_value": 1.0, 
    "ros_map_unknown_cell_value": 205, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true, 
    "tf_cache_lookup_frequency": 10, 
    "tf_cache_max_time": 10000.0
}

```

MichaelGrupp, 2021-06-25:
You need to add `--plot` to show the plot window (or `--save_plot` if you just want to save to a file directly).

kc944201, 2021-06-26:
Thanks for your contribution!
It's work for me.

closed 2021-06-26 by kc944201