MichaelGrupp/evo #413: Negative RPE?

martinakos, 2021-06-28:
In the violin plots I've done with evo, sometimes I get a part of the violin plot sticking out below zero. see below:
[image]

The violin plot represents relative error for translation with a delta of 0.5m. The actual statistics for the error are: 
```
             max        mean      median          min        rmse       sse 
0.50m   0.133578   0.0444456   0.0316042  0.000759414   0.0557532   16.0705 
```
So no negative values. It wouldn't make sense otherwise. 
Is this just a plotting artifact? can it be worked around?

MichaelGrupp, 2021-06-28:
The KDE curve of a violin plot can go beyond the actual data range. So this is just a normal thing that can happen with violin plots, but you could cut the curve by setting the `cut` parameter of seaborn to 0 (https://seaborn.pydata.org/generated/seaborn.violinplot.html)

martinakos, 2021-06-28:
The cut parameter solves the problem. Thanks.

closed 2021-06-28 by martinakos