MichaelGrupp/evo #415: Support for static tf

dan-dan97, 2021-06-30:
Hello!

While using evo with bag files I found one problem with tf (of course I mean tf2). The problem is that there is no support for static tf from topic /tf_static. Let's see my example.

We have gps sensor that publishes tf world->gps. The sensor is mounted with some translation so we have static tf gps->base_link. We want to analyze trajectory of tf world->base_link with EVO, but it is unmcomfortable because evo with bag files listenes to only topic /tf and no /tf_static, so we have to create some extra node which will listen to all tf and publish the needed data to separate topic in the form of PoseStamped topic for example.

It would be very usefull if you will add support for static tf. As you see without it usual dynamic tf isn't so comfortable in use.
This can be imlemetated through assumption that tf topics are always /tf and /tf_static so word "tf" in argument of command line interface should say that the argument is a transform between to frames, not usual topic. And the tf can be found from constant topcis /tf and /tf_static.

MichaelGrupp, 2021-06-30:
Loading TF data from bags is already possible, also with transformations that include static edges in the transform tree (provided that you have the standard topics `/tf` and `/tf_static` in the bag). For this you need to use the syntax that is also documented in the wiki - in your case it would be something like this:

```
evo_traj bag <your bag> /tf:world.base_link --verbose
```
(/tf_static is implicitly loaded)

dan-dan97, 2021-07-02:
@MichaelGrupp sorry, didn't notice that it's already supported.
Thank you very much!

closed 2021-07-02 by dan-dan97