MichaelGrupp/evo #416: Question about Relative Rotational Error per unit of angle

martinakos, 2021-06-30:
I have some trouble understanding the meaning of the relative rotational error for rotation_angle_deg and units degrees.

I have calculated the rotational error in degrees per unit of length, for a few interval lengths 0.1m, 0.2m, 0.5m 1m 2m, and plotted violin plots for them. I can understand these plots as a distribution of angular errors after having travelled the corresponding interval lengths. However, I thought whether that was a meaningful plot as the shape of the violin plot would depend more on the trajectory I'm evaluating on that on the VIO system that produced that trajectory. For example, if the trajectory has few turns and it's mostly straight lines we would get small relative rotational errors per unit of length as most length intervals would fall on straight lines. 

See example plot: 
[image]

So then I thought that what makes sense is to plot the relative rotational error in degrees per unit of angle in degrees too, for a few intervals of angles 5, 10, 45, 90, 180 degs, then I would have the same type of units in the both axes of the violin plot. This took ages to compute (compared to computing the relative translation errors per unit of length for the same trajectories) and it couldn't do the 180 deg interval as there wasn't any data points for that interval (I was surprised about this because the trajectory includes multiple loops). More fundamentally, I realise I don't understand this type of rotational error per unit of angle.  Is an interval of say 45 degs all the poses in the trajectory between two of them that differs by 45 degs? This could make the interval sizes over which the errors are computed to vary wildly in terms of how many individual poses are there in the interval (I guess this could explain why it takes so long to compute). 

See example plot:
[image]

So I'm not sure how to interpret the relative rotational error per unit of angle. And I don't know why it takes so long to compute, I'm using all_pairs=True but even though it may take the order of 1h to compute vs a couple of minutes for the same trajectories in relative translational error per unit of length. 
Also, I use `evo_res * --plot --use_filenames --ignore_title` to get the violin plots but as you can see they don't come up in the correct order. Is there any easy way to select the order of the plots?

martinakos, 2021-07-05:
Any ideas on this? or would it be better to forget about Relative Rotational Errors?

martinakos, 2021-08-18:
any help on this?!!

MichaelGrupp, 2021-08-24:
> couldn't do the 180 deg interval as there wasn't any data points for that interval

While checking the code I found a bug in the search for the pose pairs based on angle delta. It probably went unnoticed because I rarely used RPE for pairs with angle delta in the past. It's now fixed in the newest version and should work more robust. https://github.com/MichaelGrupp/evo/commit/ca85583477f930e79663af5e1e44d2b904832e66

The actual computation of angular RPE after the pairs are found is unaffected by this and works as before.

> This took ages to compute

The all-pairs search for pairs with a certain angle between them is an exhaustive search. For each pose, the rotation w.r.t. to each of the following other poses needs to be tested. This can take long to compute and create a large number of input poses for the RPE calculation.

> Is an interval of say 45 degs all the poses in the trajectory between two of them that differs by 45 degs?

In all-pairs mode, yes.

Otherwise the delta is interpreted as a threshold for the sum of the absolute angular movement accumulated along the trajectory between consecutive pairs. Aka: you start at one pose, go to the next ones until you collected enough of rotational motion; and then continue from there.

Overall, the "RPE per rotation angle" is not as trivial to explain or compute as the one based on distance, but I also don't have a better solution.

martinakos, 2021-08-25:
Thanks for the explanation. I'll repeat my tests the newest version.

closed 2021-09-13 by MichaelGrupp