MichaelGrupp/evo #417: Questions about APE and RPE

kc944201, 2021-07-01:
HI, I want to use evo_ape and evo_rpe to evaluate trajectorys with rosbag and TF, and I found the plot and data was wrong, I don't know how to resolve.
First, the reference-trajectory is /move_base/current_goal, and the estimated-trajectory is base_footprint
there are diffrene timestamps, so add --t_offset to align the timestamp, I check the evo_traj result is right.
[image]
[image]
[image]

But I use the following command, I didn't get a right result.
**Command:**
```
evo_ape bag 0626_merge.bag /move_base/current_goal base_footprint -p -v --t_offset=10650209.117645
```

**Console output:**

[image]
[image]
[image]

MichaelGrupp, 2021-07-01:
Take a look at the output of `evo_ape`. It only finds 1 matching pose, so your timestamps are probably not correctly aligned. Also if you use `evo_traj` you should use the same `--t_offset` value to make it comparable to what you are using in `evo_ape`.

Overall you need to fix your input data or compensate the timestamp offset correctly, there's not much I can do for you.

kc944201, 2021-07-01:
I added t_max_diff value with a random number, and I got the plot
[image]

also I check the data file, the timestamp and the number of poses between them isn't same.
Do you any method can align the timestamp?
Thank you!

MichaelGrupp, 2021-07-02:
As I said: you need to fix your data. Using a random `--t_max_diff` will of course not help.

> Do you any method can align the timestamp?

No, you need to find a suitable way to synchronize time in your setup yourself.

closed 2021-07-13 by MichaelGrupp