MichaelGrupp/evo #421: RPE for positions only

stefangachter, 2021-07-13:
I am using evo for evaluating VIO results with respect to GNSS reference trajectory. I do an alignment between VIO and GNSS reference, but given that GNSS provides only positions, I ignore the attitude. In particular, I assign identity as follows to create `PoseTrajectory3D` object:  `quat = np.c_[np.ones((len(stamps),1)), np.zeros((len(stamps),3))]`

I do the alignment for the first 20 poses. The alignment is successful and APE is as expected, see plot below. The problem is, though, the computation of RPE, because after alignment, the attitude of the VIO trajectory is not identity anymore. Then, computing the RPE with `rpe_base`, the relative translation of consecutive VIO trajectory chunks are rotated with respect to the ones of GNSS reference trajectory. The resulting RPE is far too large, see figure below. It is expected to be smaller than 15cm.

I am wondering, if there is better way to compute the RPE, if only positions are available for ground-truth. (One way is to do the alignment beforehand, ignoring attitude, and then passing the trajectories to `rpe` function.)

[image]
[image]
[image]
[image]

MichaelGrupp, 2021-08-23:
It would be possible to just compare the distance between positions instead of the translation part of their relative pose.

Although this can give unclear results. E.g. here the distance between poses is always 1m in both paths, although there is a clear difference.
So maybe it's better to use the absolute metric in cases where no orientation is available.

[image]
[test1.txt](https://github.com/MichaelGrupp/evo/files/7033290/test1.txt)
[test2.txt](https://github.com/MichaelGrupp/evo/files/7033291/test2.txt)

MichaelGrupp, 2021-09-04:
> I agree that alignment error impacts RPE

This not true (or only partially, for scale/translation). I assume with "alignment error" you refer to one of the alignment methods offered in evo.

The transformation done with the alignment is either a rigid or a similarity transform (if with scaling) applied to the whole trajectory in the inertial frame. So it always affects APE, which is measured in the inertial frame. RPE is not calculated in the inertial frame but between two local frames i & j on each trajectory, so it's not affected by a rigid alignment transformation in any case (and if scale is aligned, it only affects translation).

In the end it all lies in the math behind RPE:

* RPE for rotation
  * compares relative rotation from frame i to j
  * influenced by orientation of frame j w.r.t. frame i 
  * invariant to rigid & similarity transformation of trajectory in inertial frame

* RPE for translation
  * compares translation vector from i to j
  * influenced by translation vector (magnitude & direction) w.r.t. frame i
  * invariant to rigid transformation of trajectory in inertial frame

* RPE for distance only (see my previous comment, currently not implemented)
  * compares distance from frame i to j
  * influenced by translation vector magnitude only
  * invariant to rigid transformation of trajectory in inertial frame

Yes, the translation metric is not meaningful if you only have positions. The only relative measure that could be done is the distance. But the point of my previous comment was that measuring the relative distances can have ambiguous results. Besides that the distance RPE could of course be easily added to evo if there's demand for such a feature.

stefangachter, 2021-09-06:
Thanks for your further explanation. I agree with it. You are completely right. Sorry, I was stuck. RPE should be the same before and after any alignment exactly because it is a relative measure. Often, only position reference is given, such as GNSS position measurements. Thus, distance RPE in evo would be welcomed to have some measures on scale error or drift.

For verification, translation RPE without and with alignment:
[image]
[image]
The error is exactly the same.

Translation APE without and with alignment:
[image]
[image]
The error is pretty different.

Sorry again for disturbing.

closed 2021-09-06 by stefangachter

MichaelGrupp, 2021-09-06:
No problem. I just released this as a new feature that can be used with `--pose_relation point_distance` in the command line. With this, the relative error in my dummy example above is exactly zero.

stefangachter, 2021-09-07:
Thanks a lot for adding this option. For my clarification, the point distance error is defined as
[image]
if I understand your code correctly. That is the the difference of the norm of the translations from i to j.
Might it be useful to have a scale error as well? Something like:
[image]

MichaelGrupp, 2021-09-15:
Can be done, I implemented it here: https://github.com/MichaelGrupp/evo/pull/432. In my tests it behaves as expected, but feel free to also test it. It shows the ratio in percent, I think it's more useful this way.