MichaelGrupp/evo #428: How to use evo_res to create multiple boxplot with different methods and multiple deltas for RPE in one graph?

spc2, 2021-09-01:
Thanks for your great work and it helps me a lot.
 Description:
Example in the README can use evo_res draw the result comparison with multiple methods and one delta or one method with multiple deltas for RPE. However, I want to get result comparison with multiple methods and multiple deltas just as follows:

[image]

And it is possible for evo to create such result comparison?

MichaelGrupp, 2021-09-13:
This is not directly possible with evo_res. But you can create a custom script with Python that loads the data from the result files and plots them the way you want. This older answer could help: https://github.com/MichaelGrupp/evo/issues/160#issuecomment-488590818

spc2, 2021-09-14:
Thank you for your responses. I will have a try.

closed 2021-09-15 by MichaelGrupp