MichaelGrupp/evo #431: How to change the scale of axis

Benjamin-08, 2021-09-12:
Thank you so much for this excellent work which help us a lot to evaluate the precision of localization. However, when it comes to the situation where the trajectory is a long-straight line, chances are that the scale of axis in trajectory map can not be adjusted to show the difference of two lines in detail. So could you help us how to solve this problem? Our sincere thanks to you !!!

MichaelGrupp, 2021-09-13:
I'm not sure what you mean - please post an example to illustrate.

Teemoshan, 2021-09-13:
[image]
As is shown in Fig, the scale of axis reduce 0.5

Teemoshan, 2021-09-13:
@MichaelGrupp How can I change the  the scale of axis to get the right column Fig

MichaelGrupp, 2021-09-15:
@Teemoshan you can use the zoom tool in the plot window 
[image]

Teemoshan, 2021-09-16:
Ok, it can work. Thank you!

Benjamin-08, 2021-09-16:
> I'm not sure what you mean - please post an example to illustrate.
Thank you so much and I think I have solved my problem by deleting tools/plot.py line 255, which can help to set the range of axis.

closed 2021-09-23 by MichaelGrupp

majdros, 2023-07-30:
How can I adjust or refine the scale of the time on the x-axis? In my example, the time scale on the x-axis is too big (50 seconds), and I would like to have, for example, 10 seconds. If I use the zoom tool, a part of my characteristic curves disappears.
[image]