MichaelGrupp/evo #433: After evo_traj I got a fully aligned xyz trajectory but a rpy trajectory with Fixed deviation，Why?

KerunLi, 2021-09-15:
The two trajectories can almost completely coincide on the xyz axis, but there are fixed deviations in the three directions of the rpy axis. The deviation of each axis is fixed, but the deviations of the three axes are different. Why? 
Thanks for any help.
[image]
[image]

MichaelGrupp, 2021-09-15:
I would assume that the pose axes of your two trajectories are oriented differently, i.e. wrong extrinsic calibration. You can view how this looks like with https://github.com/MichaelGrupp/evo/wiki/Plotting#coordinate-axis-markers
Apart from that, nobody can help you here without knowing your data.

KerunLi, 2021-09-15:
> I would assume that the pose axes of your two trajectories are oriented differently, i.e. wrong extrinsic calibration. You can view how this looks like with https://github.com/MichaelGrupp/evo/wiki/Plotting#coordinate-axis-markers
> Apart from that, nobody can help you here without knowing your data.

[data.txt](https://github.com/MichaelGrupp/evo/files/7169295/data.txt)
[traj_mh05_sorted.txt](https://github.com/MichaelGrupp/evo/files/7169297/traj_mh05_sorted.txt)
Thanks a lot！There is my data, and data.txt is the groundtruth in euroc/MH_05_difficult, traj_mh05_sorted.txt is the result of rpg_svo_pro_open(SVO2.0), I ofstream the frame->T_f_w_.inverse() to the traj_mh05_sorted.txt. 
[image]

KerunLi, 2021-09-16:
I got it! Because the groundtruth of euroc/MH_05_difficult is T_world_body, and the T_world_imu is the traj I want to evaluate. Usually, IMU is treated as body

closed 2021-09-16 by KerunLi