MichaelGrupp/evo #436: Compare two odom with a static transform between

tomlogan501, 2021-09-24:
Hi,
As the title said, I try your software to compare 2 gps odoms, one made with an INS and the other based on the first antenna (I  think, I would like to confirm it with your software)
I try to use the option delta offset but it doesn't seem to be the use.
I also retreive the best alignement to edit it but I would like to apply an offset for each pair of pose.
I dropped a question also on the forum ros answer  : https://answers.ros.org/question/387291/compare-two-odoms-with-different-frames-with-evo/

MichaelGrupp, 2021-09-29:
If I understand correctly, you have two trajectories from two sensors that are mounted at different places on your device? Then you need to compensate for that offset based on extrinsic calibration of these sensors - (I guess this is the static offset you mentioned?).

You can transform a trajectory with an offset that is applied to each pose individually (right-multiplication instead of left / rigid transform of the whole trajectory) using `evo_traj .... --transform_right --save_as_tum` and then use this.
https://github.com/MichaelGrupp/evo/wiki/evo_traj#transformations

But there is no automatic way to do this in evo - you need to know the offset and the math needed for your sensor setup.

tomlogan501, 2021-09-30:
Thanks for your reply.
I got the transform from tf between the 2 frames. So I should convert it using evo_traj and this transform first ?
Would it be better to use evo_rpe with tf then ?

closed 2023-11-27 by MichaelGrupp