MichaelGrupp/evo #441: evo can't plot topics of Rosbag

zhaohaowu, 2021-11-11:
When I run` evo_traj  bag ROS_example.bag --all_topics -p`
there is an error, as follows:
```
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/evo/entry_points.py", line 92, in launch
    main_module.run(args)
  File "/usr/local/lib/python3.6/dist-packages/evo/main_traj.py", line 317, in run
    trajectories, ref_traj = load_trajectories(args)
  File "/usr/local/lib/python3.6/dist-packages/evo/main_traj.py", line 213, in load_trajectories
    import rosbag
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/__init__.py", line 33, in <module>
    from .bag import Bag, Compression, ROSBagException, ROSBagFormatException, ROSBagUnindexedException
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 53, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)
```
But running` evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p ` is ok

MichaelGrupp, 2021-11-11:
You have rosbag for ROS melodic (Python 2) but you installed evo for Python 3, this won't work.

closed 2021-12-02 by MichaelGrupp

SteveRogan, 2022-03-09:
So how to solve this problem？

mrtnbm, 2023-02-22:
> So how to solve this problem？

Downgrade to Python 2 version of the package or upgrade your ROS distribution.