MichaelGrupp/evo #455: Colors

femust, 2022-01-07:
Hey,

is there a way to get information regarding the sequential colors that will happen during evo_traj? I need this information to draw the same color lines of other plots that I am doing.

MichaelGrupp, 2022-01-12:
The plot_seaborn_palette config is what you are looking for:
```
evo_config show
```
https://seaborn.pydata.org/generated/seaborn.color_palette.html

femust, 2022-01-12:
is there a way to turn off this palette and have a default one?

MichaelGrupp, 2022-01-12:
If with "default" you mean the default matplotlib palette, then I think this  "tab10":
```
evo_config set plot_seaborn_palette tab10
```
You should be able to find this in the docs of matplotlib or seaborn.

femust, 2022-01-30:
Great, thanks!

closed 2022-01-30 by femust