MichaelGrupp/evo #459: Question about trajectory alignment.

Boyuan-Tian, 2022-02-04:
Hi team,

Thanks for offering such an excellent tool!

I have a question about the trajectory alignment result shown below.

In the figure, I assigned “ICP_Pose” as the reference of the other two trajectories, with the “-as -p” parameter enabled.

However, the “ICP_Pose” on the “Yaw” axis has a consistent offset from others. The dashed curve intuitively can be moved slightly higher than in the figure, which is supposed to be handled by the tool from my understanding to the alignment mechanism.

I wondered if there is any misunderstanding on my side, and do you have a sense of what could cause this issue.

I attached the three trajectory files here for reproduction, and all of them are in TUM format.

Thank you so much in advance!

Best regards,
Boyuan Tian

MichaelGrupp, 2022-02-07:
The Umeyama alignment algorithm minimizes the metric distance between all data points that are used for alignment. Rotation is not considered.

Instead of looking at the rotation in the global frame, it's usually better to look at the relative error. In your case e.g.:
```
evo_rpe tum ICP_Pose.txt illixr-ov-pose-tum.csv --pose_relation angle_deg -p
```

Boyuan-Tian, 2022-02-16:
Gotcha! Thank you so much!
Btw do you have any idea what tool could handle all 6DoFs?

MichaelGrupp, 2022-02-17:
No, unfortunately don't know something that I can recommend for that right now.