MichaelGrupp/evo #464: Problem with the evo alignment

Gatsby23, 2022-02-20:
**Description:**

Hello, thanks for your wonderful tool for evaluation. But there are some problem with the lidar trajectory output. I run A-LOAM to test the lidar trajectory. I have get the laser_odom_path and the aftermap_path. I use the evo_traj to compare the two result with the groundtruth, which is collected by the vicion. The whole process like this:
[image]
**The trajectory File**
here is the ground truth file, which collected by the vicon
[tum-cutted-m-local.txt](https://github.com/MichaelGrupp/evo/files/8104473/tum-cutted-m-local.txt)
here is the two lidar trajectory, generated from A-LOAM
[TrajAfterMapped.txt](https://github.com/MichaelGrupp/evo/files/8104474/TrajAfterMapped.txt)
[TrajLaserOdometry.txt](https://github.com/MichaelGrupp/evo/files/8104475/TrajLaserOdometry.txt)

I really don't know, why the trajectory can't be aligned. I also checked the timestamp file like this:
[image]
It shows both of the trajectories looks fine. I really like the evo tools, but don't know how to fix this problem. Could please give me some advice about this? 
My evo version is 1.18.0 and I have tested on Ubuntu 20.04 and Ubuntu18.04
Thank you very much.

MichaelGrupp, 2022-02-21:
These errors usually happen if timestamps in the files are not correctly synchronized. Are you sure the time synchronization of your two systems is correct? Because I tried your data and it gets better when a time offset correction of ~ -16 seconds is used:
```
evo_traj tum TrajAfterMapped.txt --ref tum-cutted-m-local.txt -apv --plot_relative_time --t_offset -16
```

Gatsby23, 2022-02-21:
> These errors usually happen if timestamps in the files are not correctly synchronized. Are you sure the time synchronization of your two systems is correct? Because I tried your data and it gets [...]

Thank you very much. It's weird that we have checked the timestamp problem, but there is still have the offset between the groundtruth with the collected sensors. After that, the result is better enough
[image]

MichaelGrupp, 2022-02-21:
> It's weird that we have checked the timestamp problem, but there is still have the offset

Well, the tool only checks if the timestamps in each trajectory are sorted correctly. Calibrating your system is not part of that.

Gatsby23, 2022-02-22:
> > It's weird that we have checked the timestamp problem, but there is still have the offset
> 
> Well, the tool only checks if the timestamps in each trajectory are sorted correctly. Calibrating your system is not part of that.

Thanks for your kindly reply. We use the ntp command to synchronize the timestamp between vicon and the data collecting device. But it seems there are still have some problems with the two devices. I'll check it carefully. Thank you very much.

closed 2022-02-22 by Gatsby23

YoungHaulk, 2023-05-05:
You probably got the base time of the mocap system wrong. Because there is a time delay in collecting data, the reference time for mocap is not the time when you press the record button