MichaelGrupp/evo #465: Why can't I open a .txt file with evo_traj?

J1dan, 2022-03-06:
When I input this command:
_evo_traj tum FrameTrajectory_TUM_Format.txt --plot --plot_mode=xy[ERROR]_ 
it prints
_csv file FrameTrajectory_TUM_Format.txt does not exist_
But I have seen others plotting trajectory using similar command like
_evo_traj tum data.tum CameraTrajectory.txt --plot --plot_mode=xy_
Could someone please give me some help?

MichaelGrupp, 2022-03-07:
"csv file FrameTrajectory_TUM_Format.txt does not exist" already says what the problem is: there is no `FrameTrajectory_TUM_Format.txt` in the directory in which you execute the command. You need to use a correct file path to load a file.

J1dan, 2022-03-07:
Thanks for your response. But there do have `FrameTrajectory_TUM_Format.txt` in the path. I managed to solve the problem by adjusting the code from `SLAM.SaveKeyFrameTrajectoryTUM("KeyFrameTrajectory_TUM_Format.txt");` to `SLAM.SaveKeyFrameTrajectoryTUM("KeyFrameTrajectory_TUM_Format.csv");`.
Thanks again for your suggestion.

closed 2022-03-07 by J1dan